apache/plc4x #568: (plc4j opcua driver) Unable to read when node id contains unicode characters

modraedlau, 2022-11-02:
Unable to read when node id contains unicode characters. 
For example: 
The nodeid is `ns=2;s=设备1`, the read result will be `NOT_FOUND`.

Look at the generated class `org.apache.plc4x.java.opcua.readwrite.PascalString`, the `getStringLength` equals to string length, should this be equal to `getBytes().length`?

hutcheb, 2022-11-03:
Thanks modraediau, You are right, the OPCUA standard doesn't seem to define the format of the Node Id string Identifier.

It would be great if you could create a PR for this. The PascalString class is defined in the mspec/xslt file here https://github.com/apache/plc4x/blob/develop/protocols/opcua/src/main/xslt/opc-manual.xsl

If you have any issues, or need some help let us know.

If you aren't in a position to create a PR, I'd be more than happy to test the change in the next few days.

splatch, 2022-11-03:
@modraedlau It might be related to utf-8 vs utf-16. What's your platform jvm setting in this regard?

modraedlau, 2022-11-23:
I submitted a pull request：[fix(plc4j/opcua): not to get String.length but to calculate length in...](https://github.com/apache/plc4x/pull/668)

closed 2022-11-24 by modraedlau