apache/plc4x #587: The connection pool should respect the entire URL

ottlukas, 2022-11-09:
Currently the Connection pool only seems to work on the connection string but ignoring the parameters. As the parameters might contain important information (Rack & Slot for S7 or other things like the location of the KNXProj File for KNX or EDE File locations for BACnet) it should treat the entire connection string as pool key. 

Imported from Jira [PLC4X-170](https://issues.apache.org/jira/browse/PLC4X-170). Original Jira may contain additional context.
Reported by: cdutz.

ottlukas, 2024-02-06:
@chrisdutz is this still an issue?

chrisdutz, 2024-02-06:
Well ... technically the connection pool no longer exists, but I just checked and the connection cache also uses the full URL

closed 2024-02-06 by chrisdutz