apache/plc4x #590: The RequestTransactionManagerTest.abortTransactionFromExternally seems to be a little flaky

ottlukas, 2022-11-09:
The build on jenkins randomly seems to fail because of this test:

RequestTransactionManagerTest.abortTransactionFromExternally
### Stacktrace

java.lang.AssertionError at org.apache.plc4x.java.spi.optimizer.RequestTransactionManagerTest.abortTransactionFromExternally(RequestTransactionManagerTest.java:153)

 

Imported from Jira [PLC4X-171](https://issues.apache.org/jira/browse/PLC4X-171). Original Jira may contain additional context.
Reported by: cdutz.

chrisdutz, 2024-02-06:
This test hasn't been causing any real issues ...

sruehl, 2024-02-06:
do you have a link to when it was disabled (e.g. commit)?

chrisdutz, 2024-02-06:
On the 07.02.2020 :-)

chrisdutz, 2024-02-06:
a811636c4b0c17aff7caab1f825dbd94bae9fac0

chrisdutz, 2024-02-10:
I re-enabled it, and we'll investigate it, if it starts causing problems again.

commit b2f30fa4 (2024-02-10): chore: Re enabled a test in order to see, if it's really flaky

fixes: #590
closed 2024-02-10 by chrisdutz