apache/plc4x #591: Request split / Message Split

ottlukas, 2022-11-09:
- A request with many Items is divided within the limits allowed by the size of the PDU.
 - If one of the requested Items exceeds the size of the PDU it is trimmed to the maximum size of the PDU. The existing code tries to split the message, but it fails. This generates an unsafe condition.

 - Rethink the routine for handling long messages, as an additional layer in Netty.

Imported from Jira [PLC4X-182](https://issues.apache.org/jira/browse/PLC4X-182). Original Jira may contain additional context.
Reported by: cgarcia.

chrisdutz, 2024-02-10:
@glcj I think you also implemented this, right? Can this issue be closed?

glcj, 2024-02-12:
Hello,

This routine is implemented in the S7Optimizer, I have carried out tests with the default PDU which is something around 400 bytes and I have not found any problems.

For my part, this i ssue can be considered closed.

chrisdutz, 2024-06-30:
Related to #605

closed 2024-10-18 by chrisdutz