apache/plc4x #592: Subscription to Items

ottlukas, 2022-11-09:
This version implements the subscription to PLC Items under the philosophy "Don't call me, I'll call you".
 The requested items are sent periodically by the PLC to the driver. The driver collects the data and sends it to the client.
 This solution is ideal for systems that need rapid data updating and low resource consumption.

Currently only S7-300 and S7-400.
S7-1500 the next revision.

Imported from Jira [PLC4X-184](https://issues.apache.org/jira/browse/PLC4X-184). Original Jira may contain additional context.
Reported by: cgarcia.

chrisdutz, 2024-02-06:
@glcj  can you confirm this is now implemented? If yes, please close this issue.

glcj, 2024-02-12:
Hello,
In PR #1400 the observations of the work team in the implementation of this feature were incorporated.

**CYC:** Cyclic subscription to an item pull

This feature only applies to S7-300, S7-400 and VIPA devices (to be tested).

Tested with an S7-400. I am waiting for a CP 343-1 for additional testing, but I consider the issue closed.

closed 2024-02-12 by glcj