apache/plc4x #596: User manual. Driver-S7

ottlukas, 2022-11-09:
The objective of the manual is to show in detail the characteristics of the developed Driver-S7, its potential and points of improvement.
It will be done incapie in the formats of items and in the best practices to optimize the communication process with Siemens S7 PLCs.
It will end with practical examples associated with Continuous (machinery) and Batch (Processes) processes.

Imported from Jira [PLC4X-180](https://issues.apache.org/jira/browse/PLC4X-180). Original Jira may contain additional context.
Reported by: cgarcia.

chrisdutz, 2024-02-06:
@glcj I think this issue can be closed, right? If yes, please close it.

glcj, 2024-02-07:
Many of the driver's features are documented on the page. The update will be done directly on the website.

closed 2024-02-07 by glcj