apache/plc4x #602: Pending threads after connection.close

ottlukas, 2022-11-09:
I’m facing some issue when I try to quit my application. It seems connection.close() is not stopping all pending threads. Is this a known issue or am I doing something wrong?

```

String url = "s7://...";
PlcDriverManager manager = new PlcDriverManager();
PlcConnection connection
= manager.getConnection(url);
connection.close();
System.out.println("closed”);  // gets printed but
hangs afterwards
```

I created a thread dump which is attached.

Imported from Jira [PLC4X-249](https://issues.apache.org/jira/browse/PLC4X-249). Original Jira may contain additional context.
Reported by: svoss.

chrisdutz, 2024-02-10:
I have checked this for most drivers and these zombie threads seem to have been fixed.

closed 2024-02-10 by chrisdutz

glcj, 2024-02-11:
I made some improvements in version 0.6.1. The one you are running is the default one and so far I have not had any problems. I think it can be closed. +1, close