apache/plc4x #604: [Modbus] Apache NiFi processor throws java.io.IOException after a while

ottlukas, 2022-11-09:
My Plc4xSourceProcessor's, PLC connection String is "modbus:tcp://[10.0.2.238:502?slave=1](http://10.0.2.238:502/?slave=1)" and PLC resource address String is "test1=holding-register:1"
  
 I can get the values for 5-10 times with 5 second intervals but then I get below exception. I can read the values with the Modbus Poll application, so most probably the PLC4X side has a problem.
  
 I may also get some other exceptions when starting the processor, which are also below.
  
PS: Wireshark trace is attached. I've read 16 times then I get the exception.
  
 * 
 \-- This is the exception which I get after some successful read operations.*
 2020-08-27 13:19:06,091 WARN [nioEventLoopGroup-8-1] io.netty.channel.DefaultChannelPipeline An exceptionCaught() event was fired, and it reached at the tail of the pipeline. It usually means the last handler in the pipeline did not handle the exception.
 java.io.IOException: An existing connection was forcibly closed by the remote host
 at sun.nio.ch.SocketDispatcher.read0(Native Method)
 at sun.nio.ch.SocketDispatcher.read(Unknown Source)
 at sun.nio.ch.IOUtil.readIntoNativeBuffer(Unknown Source)
 at sun.nio.ch.IOUtil.read(Unknown Source)
 at sun.nio.ch.SocketChannelImpl.read(Unknown Source)
 at io.netty.buffer.PooledByteBuf.setBytes(PooledByteBuf.java:253)
 at io.netty.buffer.AbstractByteBuf.writeBytes(AbstractByteBuf.java:1133)
 at io.netty.channel.socket.nio.NioSocketChannel.doReadBytes(NioSocketChannel.java:350)
 at io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:148)
 at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:714)
 at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:650)
 at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:576)
 at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:493)
 at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:989)
 at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
 at io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
 at java.lang.Thread.run(Unknown Source)
  
  
 \-- This is the exception which I get sometimes when starting up the processor.*
 2020-08-27 13:17:16,813 WARN [Timer-Driven Process Thread-11] o.a.n.controller.tasks.ConnectableTask Administratively Yielding Plc4xSourceProcessor[id=2f12f5b4-0174-1000-724e-a53ba0fc1652] due to uncaught Exception: java.lang.NullPointerException
 java.lang.NullPointerException: null
 at org.apache.plc4x.nifi.Plc4xSourceProcessor.onTrigger(Plc4xSourceProcessor.java:50)
 at org.apache.nifi.processor.AbstractProcessor.onTrigger(AbstractProcessor.java:27)
 at org.apache.nifi.controller.StandardProcessorNode.onTrigger(StandardProcessorNode.java:1174)
 at org.apache.nifi.controller.tasks.ConnectableTask.invoke(ConnectableTask.java:213)
 at org.apache.nifi.controller.scheduling.TimerDrivenSchedulingAgent$1.run(TimerDrivenSchedulingAgent.java:117)
 at org.apache.nifi.engine.FlowEngine$2.run(FlowEngine.java:110)
 at java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
 at java.util.concurrent.FutureTask.runAndReset(Unknown Source)
 at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(Unknown Source)
 at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
 at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
 at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
 at java.lang.Thread.run(Unknown Source)

Imported from Jira [PLC4X-245](https://issues.apache.org/jira/browse/PLC4X-245). Original Jira may contain additional context.
Reported by: turker.tunali.

chrisdutz, 2024-02-06:
Is this still an issue? I think not?

ottlukas, 2024-02-06:
Yeah we can close until further notice.

closed 2024-02-06 by ottlukas