apache/plc4x #611: Mock driver always writes null value

ottlukas, 2022-11-09:
The Mock driver is usefull for developing applications without available hardware. Reads from Mock devices work, but writes seem broken.

Scenario:

implement *MockDevice* interface and the *write* method. See that this device is set in the connection using the *MockConnection*.*setDevice* method.

When a WriteRequest is issued on the connection, the *MockDevice*.*write* method is indeed called, but the *value* parameter is always _*null*_. The given value to write is not passed through.

Debugging and looking into the sourcecode shows following:

*MockConnection.write* uses following statement to retrieve the value:

 
```

// MockConnection:147
((MockField) writeRequest.getField(name)).getPlcValue()

```

This retrieves the *plcValue* member from the MockField object. This  *plcValue* is written to by constructor *MockField(String address, MockPlcValue plcValue)* only. However, I did not find any call to this constructor.

 

Note:

Other Drivers seem to use
```

writeRequest.getPlcValue(fieldName) 

```

instead and do not have a value member in the Field class. (e.g. Simulator, Modbus )

Imported from Jira [PLC4X-288](https://issues.apache.org/jira/browse/PLC4X-288). Original Jira may contain additional context.
Reported by: teslanet-nl.

ottlukas, 2024-02-06:
@chrisdutz  Our tests are running, so can this issue be closed now?

chrisdutz, 2024-02-06:
Didn't close it, as I actually haven't tested it ... I did close an issue as it wasn't causing issues, till I noticed it wasn't as I disabled the test ... so wanted to make sure.

closed 2024-06-30 by chrisdutz