apache/plc4x #615: Docker build not working - mvnw permission denied etc

ottlukas, 2022-11-09:
Hello, I'm trying to install PLC4X with Docker and get a couple of errors - 

$ docker build -t plc4x .
...
Step 21/39 : COPY . /ws/
 \---\> e86e4d250f0c
Step 22/39 : WORKDIR /ws
 \---\> Running in b2d9525763e0
Removing intermediate container b2d9525763e0
 \---\> 2d714dffc27b
Step 23/39 : RUN ./mvnw -P with-boost,with-c,with-cpp,with-dotnet,with-go,with-logstash,with-opcua-werver,with-proxies,with-python,with-logstash,with-sandbox com.offbytwo.maven.plugins:maven-dependency-plugin:3.1.1.MDEP568:go-offline -DexcludeGroupIds=org.apache.plc4x,org.apache.plc4x.examples,org.apache.plc4x.sandbox
 \---\> Running in 97e09acbcdfd
/bin/sh: 1: ./mvnw: Permission denied
The command '/bin/sh -c ./mvnw -P with-boost,with-c,with-cpp,with-dotnet,with-go,with-logstash,with-opcua-werver,with-proxies,with-python,with-logstash,with-sandbox com.offbytwo.maven.plugins:maven-dependency-plugin:3.1.1.MDEP568:go-offline -DexcludeGroupIds=org.apache.plc4x,org.apache.plc4x.examples,org.apache.plc4x.sandbox' returned a non-zero code: 126

So I added a step to the Dockerfile line 72
 
RUN chmod ****x ./mvnw
That got it past that error, then got another one -

Non-resolvable parent POM for org.apache.plc4x.sandbox:plc4cpp:[unknown-version]: Could not find artifact org.apache.plc4x.sandbox:plc4x-sandbox:pom:0.8.0-SNAPSHOT and 'parent.relativePath' points at wrong local POM @ line 24, column 11

I'm not familiar with maven so am not sure how to get past it...

Thank you for any help - 

 

Imported from Jira [PLC4X-284](https://issues.apache.org/jira/browse/PLC4X-284). Original Jira may contain additional context.
Reported by: bburnskm.

maneetgoyal, 2023-05-14:
```
 => [build 20/24] RUN dos2unix .mvn/wrapper/maven-wrapper.properties       0.3s
 => [build 21/24] RUN ./mvnw -P with-c,with-dotnet,with-go,with-sandbo  3033.2s
 => ERROR [build 22/24] RUN ./mvnw -P with-c,with-dotnet,with-go,with-san  6.6s
------
 > [build 22/24] RUN ./mvnw -P with-c,with-dotnet,with-go,with-sandbox install:
#0 1.366 [INFO] Scanning for projects...
#0 2.941 [INFO] ------------------------------------------------------------------------
#0 2.943 [INFO] Reactor Build Order:
#0 2.944 [INFO] 
#0 2.946 [INFO] PLC4X                                                              [pom]
#0 2.949 [INFO] Code-Generation                                                    [pom]
#0 2.953 [INFO] Code-Generation: Protocol Base: MSpec                              [jar]
#0 2.956 [INFO] Code-Generation: Language Base: Freemarker                         [jar]
#0 2.959 [INFO] PLC4J                                                              [pom]
#0 2.961 [INFO] PLC4J: API                                                         [jar]
#0 2.963 [INFO] PLC4J: SPI                                                         [jar]
#0 2.965 [INFO] PLC4J: Transports                                                  [pom]
#0 2.968 [INFO] PLC4J: Transports: TCP                                             [jar]
#0 2.970 [INFO] PLC4J: Transports: Test                                            [jar]
#0 2.972 [INFO] PLC4J: Utils                                                       [pom]
#0 2.974 [INFO] PLC4J: Utils: Test Utils                                           [jar]
#0 2.975 [INFO] Code-Generation: Protocol Test: Test Mspec                         [jar]
#0 2.976 [INFO] Code-Generation: Language: Java                                    [jar]
#0 2.977 [INFO] Code-Generation: Language: C                                       [jar]
#0 2.978 [INFO] Code-Generation: Language: C#                                      [jar]
#0 2.979 [INFO] Code-Generation: Language: Go                                      [jar]
#0 2.980 [INFO] Protocols                                                          [pom]
#0 2.981 [INFO] Protocols: AB-ETH                                                  [jar]
#0 2.982 [INFO] Protocols: ADS                                                     [jar]
#0 2.983 [INFO] Protocols: C-Bus                                                   [jar]
#0 2.984 [INFO] Protocols: BACNet/IP                                               [jar]
#0 2.985 [INFO] Protocols: CANopen                                                 [jar]
#0 2.986 [INFO] Protocols: Df1                                                     [jar]
#0 2.987 [INFO] Protocols: EtherNet/IP                                             [jar]
#0 2.987 [INFO] Protocols: Firmata                                                 [jar]
#0 2.988 [INFO] Protocols: Generic CAN                                             [jar]
#0 2.989 [INFO] Protocols: KNXNet/IP                                               [jar]
#0 2.990 [INFO] Protocols: Modbus                                                  [jar]
#0 2.991 [INFO] Protocols: MQTT                                                    [jar]
#0 2.992 [INFO] Protocols: OPC UA                                                  [jar]
#0 2.993 [INFO] Protocols: Open-Protocol                                           [jar]
#0 2.994 [INFO] Protocols: PLC4X (Proxy)                                           [jar]
#0 2.996 [INFO] Protocols: PLC4X (API)                                             [jar]
#0 2.997 [INFO] Protocols: Profinet                                                [jar]
#0 2.998 [INFO] Protocols: S7                                                      [jar]
#0 2.999 [INFO] Protocols: Simulated                                               [jar]
#0 2.999 [INFO] Protocols: SocketCAN                                               [jar]
#0 3.000 [INFO] PLC4J: Drivers                                                     [pom]
#0 3.000 [INFO] PLC4J: Driver: AB-ETH                                              [jar]
#0 3.001 [INFO] PLC4J: Transports: Serial                                          [jar]
#0 3.001 [INFO] PLC4J: Driver: Beckhoff TwinCat ADS Driver                         [jar]
#0 3.002 [INFO] PLC4J: Transports: UDP                                             [jar]
#0 3.002 [INFO] PLC4J: Utils: Pcap (Shared)                                        [jar]
#0 3.002 [INFO] PLC4J: Transports: PCAP (shared)                                   [jar]
#0 3.003 [INFO] PLC4J: Utils: Raw-Sockets                                          [jar]
#0 3.003 [INFO] PLC4J: Transports: Raw Socket                                      [jar]
#0 3.003 [INFO] PLC4J: Utils: Pcap-Replay                                          [jar]
#0 3.004 [INFO] PLC4J: Transports: PCAP Replay                                     [jar]
#0 3.004 [INFO] PLC4J: Driver: S7 (Step7)                                          [jar]
#0 3.005 [INFO] PLC4J: Utils: Test-Generator                                       [jar]
#0 3.005 [INFO] PLC4J: Driver: BACNet/IP Driver                                    [jar]
#0 3.005 [INFO] PLC4J: Driver: C-Bus                                               [jar]
#0 3.006 [INFO] PLC4J: Transports: CAN                                             [jar]
#0 3.006 [INFO] PLC4J: Transports: SocketCAN                                       [jar]
#0 3.006 [INFO] PLC4J: Transports: VirtualCAN                                      [jar]
[... cut ...] container.local_folder=/Users/maneet/Documents/plc4x --log-level debug --log-format json --config /Users/maneet/Documents/plc4x/.devcontainer/devcontainer.json --default-user-env-probe loginInteractiveShell --mount type=volume,source=vscode,target=/vscode,external=true --skip-post-create --update-remote-user-uid-default on --mount-workspace-git-root true
[3053517 ms] Exit code 1
```

Getting the above error when I try to create a Docker dev environment using the Dockerfile in the root directory.

maneetgoyal, 2023-05-17:
@sruehl @chrisdutz Any suggestions please? Hard to get the build going via Docker. I can share more details if needed.

sruehl, 2023-05-17:
Seems that the build doesn't detect dotnet anymore. @chrisdutz any idea?

sruehl, 2023-05-17:
I'll add a ouput in case that happens so It is easier to debug the problem...

sruehl, 2023-05-17:
@maneetgoyal can you retry with the latest develop revision? 62478b88010805d8d718f4f7c151207e20f7cb2e

maneetgoyal, 2023-05-17:
> @maneetgoyal can you retry with the latest develop revision? [62478b8](https://github.com/apache/plc4x/commit/62478b88010805d8d718f4f7c151207e20f7cb2e)

Sure, will try in a bit

closed 2023-05-17 by sruehl