apache/plc4x #619: [S7] Writing byte array not working

ottlukas, 2022-11-09:
I'm having some issues with writing a byte array using S7 driver to the S7-1200.

I'm a bit new to the communication with PLCs, so I'm not sure if I'm doing something wrong.

I've followed the example of how to write a byte array to the connection, however I'm getting an INTERNAL_ERROR as a response code from the library and in the Wireshark the status from the PLC is inconsistent data type. 

I've attached both the sample project that's not working for me, my definition of byte array on the PLC and also  Wireshark capture. 

It's possible that I'm not doing something right, however I can't figure it out. 

Writing of most of the single values is working, however the array is the problem. And as far as I understood from the code and documentation arrays should be supported. 

Any help would be appreciated.

Imported from Jira [PLC4X-309](https://issues.apache.org/jira/browse/PLC4X-309). Original Jira may contain additional context.
Reported by: maidab.

chrisdutz, 2024-05-17:
Ok ... I was able to confirm the issue and to resolve it in my setup ... could you please try the current SNAPSHOT version and verify if this issue is now fixed?

chrisdutz, 2024-05-17:
Just saw this issue was imported from Jira so I can't really expect any feedback ... will therefore close it right now. I have a test in my ManualTest for S7 that reads and writes byte-arrays.

closed 2024-05-17 by chrisdutz