apache/plc4x #620: Plc4x support for RTU/Serial communication for modbus protocol.

ottlukas, 2022-11-09:
Hi,

   I am using plc4x for one of our projects to develop a module which uses plc4x to establish communication with a sensor device using modbus RTU/Serial communication.

I am able to communicate with the device with TCP transport but not able to do with RTU/Serial communication.

I tried to look for some example/documentation on how to use RTU/Serial communication in plc4x but could not find any information in the website or the web.

Also looking at the documentation at [https://plc4x.apache.org/users/protocols/modbus.html](https://plc4x.apache.org/users/protocols/modbus.html) it is not clear whether modbus supports serial transport as it only lists tcp and udp.

 
|Compatible Transports:| * `tcp` (Default Port: 502)
 * `udp` (Default Port: 502)|

Kindly request to clarify if serial is supported in modbus? if so could you please point me to an example/documentation which shall be used to understand how to use it for serial communication.

Anyways I tried to use the library/driver plc4j-transport-serial  [https://plc4x.apache.org/users/transports/serial.html](https://plc4x.apache.org/users/transports/serial.html)

to communicate with device in which the connection is established but it fails to read the data with following WANING.

 

`2021-06-03-18:13:51.814 [nioEventLoopGroup-2-1] WARN  io.netty.channel.nio.NioEventLoop - Selector.select() returned prematurely 512 times in a row; rebuilding Selector org.apache.plc4x.java.transport.serial.SerialPollingSelector@28ecdc0d.`
`2021-06-03-18:13:59.630 [nioEventLoopGroup-2-1] WARN  io.netty.channel.nio.NioEventLoop - Selector.select() returned prematurely 512 times in a row; rebuilding Selector org.apache.plc4x.java.transport.serial.SerialPollingSelector@11c9a1fa.`

`My Sample code is as follows.`

 

` private void plcRtuReader() {`
` // unit-identifier=1&`
` String connectionString =`
` "modbus:serial://COM5?unit-identifier=1&baudRate=19200&stopBits=" + SerialPort.ONE_STOP_BIT`
` + "&parityBits="`
` + SerialPort.NO_PARITY + "&dataBits=8";`
` System.out.println("URL:" + connectionString);`
` try (PlcConnection plcConnection = new PlcDriverManager().getConnection(connectionString)) {`

` if (!plcConnection.getMetadata().canRead()) {`
` System.out.println("This connection doesn't support reading.");`
` return;`
` `}

` PlcReadRequest.Builder builder = plcConnection.readRequestBuilder();`
` builder.addItem("value-2", "input-register:1[2]");`
`PlcReadRequest readRequest = builder.build();`

`// CompletableFuture<? extends PlcReadResponse> asyncResponse = readRequest.execute();`
` PlcReadResponse response = readRequest.execute().get();`
` for (String fieldName : response.getFieldNames()) {`
` if (response.getResponseCode(fieldName) == PlcResponseCode.OK) {`
` int numValues = response.getNumberOfValues(fieldName);`
` // If it's just one element, output just one single line.`
` if (numValues == 1) {`
` System.out.println("Value[" + fieldName + "]: " + response.getObject(fieldName));`
` `}
` // If it's more than one element, output each in a single row.`
` else {`
` System.out.println("Value[" + fieldName + "]:");`
` for (int i = 0; i < numValues; i++) {`
` System.out.println(" - " + response.getObject(fieldName, i));`
` `}
` `}
` `}
` // Something went wrong, to output an error message instead.`
` else {`
` System.out.println(`
` "Error[" + fieldName + "]: " + response.getResponseCode(fieldName).name());`
` `}
` `}

` System.exit(0);`
` } catch (PlcConnectionException e) {`
` e.printStackTrace();`
` } catch (Exception e) {`
` e.printStackTrace();`
` `}
` `}

Thanks a lot for your help.

Regards,
Purushotham

Imported from Jira [PLC4X-300](https://issues.apache.org/jira/browse/PLC4X-300). Original Jira may contain additional context.
Reported by: psham81.

closed 2024-02-10 by chrisdutz

NdiayeeCheikh, 2025-05-09:
I have the same problem  and still searching for an example

NdiayeeCheikh, 2025-05-09:
> Hi,
>
>    I am using plc4x for one of our projects to develop a module which uses plc4x to establish communication with a sensor device using modbus RTU/Serial communication.
>
> I am able to communicate [...]

Hi, can i take a look on your tcp one ?