apache/plc4x #623: Modbus TCP Timeout not working. NIFI Processor Task gets stuck

ottlukas, 2022-11-09:
When using V0.9.0 in PLC4x NIFI-Processor for Modbus TCP, the Processor Task gets stuck, if the network connection is broken. The default timeout and the request-timeout option do not seem to have any affect on this behaviour.

If the NIFI-Processor is started, when the network connection is already broken, a timeout error is thrown. If the connection breaks after a successful initial connection could be made, the running Task gets stuck on the next Modbus TCP Request and has to be forced to terminate.

Using V0.8.0 is working fine in the same scenario with same config in NIFI. In Production, V0.9.0 with Modbus TCP was not usable, since connections broke every day. We had to revert to V0.8.0. This was tested with the PLC4x Build from [https://search.maven.org/search?q=plc4j-nifi-plc4x-nar](https://search.maven.org/search?q=plc4j-nifi-plc4x-nar) and the official latest NIFI (V1.14.0) Docker and Windows Version 

Imported from Jira [PLC4X-321](https://issues.apache.org/jira/browse/PLC4X-321). Original Jira may contain additional context.
Reported by: zedman.

zhouzaihang, 2022-11-15:
+1.
plc4go has same bug.

zjee, 2023-02-18:
same to me,
Plc4xSourceProcessor 0.10.0
nifi-1.20.0

chrisdutz, 2024-08-02:
A lot has changed, is this still an issue? I know much work has been put in some timeout handling in the last 2 years.
If I don't hear otherwise, I will close this issue the next time I come across it.

chrisdutz, 2024-08-16:
As this hasn not been re-reported and it's been open for a really long time, I assume that it has been resolved.

closed 2024-08-16 by chrisdutz