apache/plc4x #625: Class casts in new opcua driver

ottlukas, 2022-11-09:
Issue reported on mailing lists:

```

[nioEventLoopGroup-2-1] WARN io.netty.channel.DefaultChannelPipeline - An exceptionCaught() event was
fired, and it reached at the tail of the pipeline. It usually means the last handler in the pipeline
did not handle the exception.
io.netty.handler.codec.DecoderException: java.lang.ClassCastException:
class org.apache.plc4x.java.opcua.readwrite.ServiceFault cannot be cast to class org.apache.plc4x.java.opcua.readwrite.ReadResponse
(org.apache.plc4x.java.opcua.readwrite.ServiceFault and org.apache.plc4x.java.opcua.readwrite.ReadResponse
are in unnamed module of loader 'app')
        at io.netty.handler.codec.MessageToMessageDecoder.channelRead(MessageToMessageDecoder.java:98)

       at io.netty.handler.codec.MessageToMessageCodec.channelRead(MessageToMessageCodec.java:111)

       at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)

       at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)

       at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:357)

       at io.netty.handler.codec.ByteToMessageDecoder.fireChannelRead(ByteToMessageDecoder.java:324)

       at io.netty.handler.codec.ByteToMessageDecoder.channelRead(ByteToMessageDecoder.java:296)
 
      at io.netty.handler.codec.ByteToMessageCodec.channelRead(ByteToMessageCodec.java:103)
      
 at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)

       at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)

       at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:357)

       at io.netty.channel.DefaultChannelPipeline$HeadContext.channelRead(DefaultChannelPipeline.java:1410)

       at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)

       at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)

       at io.netty.channel.DefaultChannelPipeline.fireChannelRead(DefaultChannelPipeline.java:919)

       at io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:166)

       at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:719)
        at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:655)

       at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:581)
        at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:493)

       at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:986)

       at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
        at io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)

       at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: java.lang.ClassCastException:
class org.apache.plc4x.java.opcua.readwrite.ServiceFault cannot be cast to class org.apache.plc4x.java.opcua.readwrite.ReadResponse
(org.apache.plc4x.java.opcua.readwrite.ServiceFault and org.apache.plc4x.java.opcua.readwrite.ReadResponse
are in unnamed module of loader 'app')
        at org.apache.plc4x.java.opcua.protocol.OpcuaProtocolLogic.lambda$read$0(OpcuaProtocolLogic.java:177)

       at org.apache.plc4x.java.opcua.context.SecureChannel.lambda$4(SecureChannel.java:212)
     
  at org.apache.plc4x.java.spi.Plc4xNettyWrapper.decode(Plc4xNettyWrapper.java:175)
        at io.netty.handler.codec.MessageToMessageCodec$2.decode(MessageToMessageCodec.java:81)

       at io.netty.handler.codec.MessageToMessageDecoder.channelRead(MessageToMessageDecoder.java:88)

       ... 23 more

```

Imported from Jira [PLC4X-319](https://issues.apache.org/jira/browse/PLC4X-319). Original Jira may contain additional context.
Reported by: ldywicki.

splatch, 2023-12-18:
Its going to be addressed within #1007.

chrisdutz, 2024-02-06:
Can this be closed now? @splatch @hutcheb