apache/plc4x #626: java.lang.ClassCastException: DefaultPlcSubscriptionField cannot be cast to class OpcuaField

ottlukas, 2022-11-09:
Hi team -

i had followed the [https://github.com/apache/plc4x/tree/rel/0.8/plc4j/examples/hello-world-plc4x-subscription](https://github.com/apache/plc4x/tree/rel/0.8/plc4j/examples/hello-world-plc4x-subscription) and facing the below error .

version : plc4j-driver-opcua-0.8.0

java : 1.8

Exception in thread "main" java.util.concurrent.ExecutionException: java.lang.ClassCastException: org.apache.plc4x.java.spi.model.DefaultPlcSubscriptionField cannot be cast to org.apache.plc4x.java.opcua.protocol.OpcuaField

 

please kindly provide your suggestions.

Imported from Jira [PLC4X-313](https://issues.apache.org/jira/browse/PLC4X-313). Original Jira may contain additional context.
Reported by: karacc.

dremsol, 2022-11-18:
I have a similar error using `0.10.0` and S7 driver
```
[main] INFO org.apache.plc4x.java.PlcDriverManager - Instantiating new PLC Driver Manager with class loader jdk.internal.loader.ClassLoaders$AppClassLoader@5cb0d902
[main] INFO org.apache.plc4x.java.PlcDriverManager - Registering available drivers...
[main] INFO org.apache.plc4x.java.PlcDriverManager - Registering driver for Protocol s7 (Siemens S7 (Basic))
[main] INFO org.apache.plc4x.java.transport.tcp.TcpChannelFactory - Configuring Bootstrap with Configuration{local-rack=1, local-slot=1, local-tsap=0, remote-rack=0, remote-slot=0, remote-tsap=0, pduSize=1024, maxAmqCaller=8, maxAmqCallee=8, controllerType='null'}
[nioEventLoopGroup-2-1] INFO org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic - S7 Driver running in ACTIVE mode.
[main] INFO org.apache.plc4x.java.spi.connection.NettyChannelFactory - Channel is closed, closing worker Group also
EventDispacher Bye!
ObjectProcessor Bye!
[main] INFO org.apache.plc4x.java.spi.connection.NettyChannelFactory - Worker Group was closed successfully!
Exception in thread "main" java.lang.ClassCastException: class org.apache.plc4x.java.s7.readwrite.field.S7Field cannot be cast to class org.apache.plc4x.java.s7.readwrite.field.S7SubscriptionField (org.apache.plc4x.java.s7.readwrite.field.S7Field and org.apache.plc4x.java.s7.readwrite.field.S7SubscriptionField are in unnamed module of loader 'app')
        at org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.subscribe(S7ProtocolLogic.java:341)
        at org.apache.plc4x.java.spi.optimizer.BaseOptimizer.lambda$optimizedSubscribe$4(BaseOptimizer.java:120)
        at org.apache.plc4x.java.spi.optimizer.BaseOptimizer.send(BaseOptimizer.java:138)
        at org.apache.plc4x.java.spi.optimizer.BaseOptimizer.optimizedSubscribe(BaseOptimizer.java:120)
        at org.apache.plc4x.java.spi.connection.AbstractPlcConnection.subscribe(AbstractPlcConnection.java:183)
        at org.apache.plc4x.java.spi.messages.DefaultPlcSubscriptionRequest.execute(DefaultPlcSubscriptionRequest.java:66)
        at io.drebble.playground.Plc4xSubscription.run(Plc4xSubscription.java:39)
        at io.drebble.playground.Plc4xSubscription.main(Plc4xSubscription.java:68)
```

splatch, 2023-12-18:
I think this issue might be improperly labelled - it does indicate opcua, but error seems to be related to s7 protocol.

chrisdutz, 2023-12-19:
Also ist 0.8.0 really old and we changed a lot in the API since then. We're happy to help, but please use an updated version, we simply can't support every version we released.

closed 2024-02-06 by chrisdutz