apache/plc4x #627: Modbus - Kafka does not close the connections

ottlukas, 2022-11-09:
Hello, 

I have been testing the connector and Kafka connects for a few weeks now. To do these tests, I try to ingest measurements from a sensor using Modbus, and this measurement ingestion is done correctly. The problem arises when I decide to remove the connector, as this connection is not closed by default with my sensor, which leads me to serious problems (due to the limitation of devices that read by Modbus, limited to 4 users per manufacture). The only way I can close these connections is to restart the kafka connect container. Is there any option or way to force these connections to close? Additionally, here are the steps to replicate this error: 

1. Initially, I don't have any connector launched, so I don't have any active connection: active: 0, waiting: 0

2. Then, I launch the connector using Kafka Rest API. At this moment, we're importing data into the kafka cluster. Using curl, we can see that the connector is working: curl -X 'GET' http://localhost:18083/connectors/ -\> ["modbus-office"]. We can see right now that we only have 1 active connection: active: 1, waiting: 0

3. I delete the kafka connector using the rest API: curl -X 'DELETE' http://localhost:18083/connectors/modbus-office. So, now, using the previous command from the previous point, we don't see any active connector, but, analysing the active connections from the sensor, it can be seen that there is one active connection: active: 1, waiting: 0. 

This is what is giving me problems, as in theory, there should not be any active connections at the moment. We have made a proxy to limit the number of active connections, but so far, we have not been able to close it manually using the REST API of Kafka or some configuration of the connector. 

Imported from Jira [PLC4X-322](https://issues.apache.org/jira/browse/PLC4X-322). Original Jira may contain additional context.
Reported by: fdorado.

FernandoDorado, 2022-11-11:
Hello Ben, 

Thanks a lot for moving forward with this issue. However, I'm not working in Idener (where we have this problem) right now. My colleagues @lidiapb and @DanielGarciaGuirao are still working on this, in case they can take a look at this problem again

chrisdutz, 2024-02-06:
Closing this as stale issue as I guess we would have heard more of this, if the issue was still existing.

closed 2024-02-06 by chrisdutz