apache/plc4x #628: Connection to United Manufacturing Hub

ottlukas, 2022-11-09:
Hi everyone!

My name is Jeremy and I am the lead developer of the United Manufacturing Hub.

The United Manufacturing Hub is an open-source industrial IoT and manufacturing application platform enabling users to connect, store, and access all relevant data sources in industrial manufacturing sites and build user-centric dashboards and applications.

It would be really great if we could make a connection between PLC4X and the United Manufacturing Hub, e.g., in form of a configurable microservice that automatically takes data from various PLCs and pushes them into a MQTT broker. The United Manufacturing Hub (incl. Node-RED, Grafana, timescaleDB, VerneMQ) then provides the infrastructure and data models to contextualize the data and enable use-cases like OEE, Performance Management, Machine to Machine Communication, Digital Shadow, etc. Furthermore, the United Manufacturing Hub allows extracting data from other data sources as well like sensors, barcodereader or cameras.

Further information can be found here:
 * [https://docs.umh.app/docs/](https://docs.umh.app/docs/)
 * [https://github.com/united-manufacturing-hub/united-manufacturing-hub](https://github.com/united-manufacturing-hub/united-manufacturing-hub)

Looking forward to discussing how we can combine both open-source projects the most effective way.

Regards,
Jeremy

Imported from Jira [PLC4X-315](https://issues.apache.org/jira/browse/PLC4X-315). Original Jira may contain additional context.
Reported by: JeremyTheocharis.

chrisdutz, 2024-02-06:
Is there still anything needed, or can this issue be closed?

sruehl, 2024-02-07:
@JeremyTheocharis guess the question might be adressed to you

chrisdutz, 2024-02-08:
Well we do have Documentation and Examples ... even if the documentation might be slightly off every now and then, the examples always represent the current state.