apache/plc4x #631: The transaction manager executor does not allow closing connections to devices.

ottlukas, 2022-11-09:
- Identification of the executor and do not use the generic name. 
 -  Add a "shutdown" method to terminate the task executor in a controlled manner. 
 -  Remove the "static" property from the executor, so it can be used in a dynamic environment. 
 -  Modify the use of logging in derived tasks.
 - A general rule must be followed that all drivers must call the new transaction handler "shutdown" method in their respective "close" method.

Imported from Jira [PLC4X-331](https://issues.apache.org/jira/browse/PLC4X-331). Original Jira may contain additional context.
Reported by: cgarcia.

chrisdutz, 2024-02-06:
@glcj Can this issue be closed?

chrisdutz, 2024-08-16:
Ping? Next time I come across this issue, I will close it. (Seeing it's imported from Jira also shows it's age)

closed 2024-08-30 by chrisdutz