apache/plc4x #639: s7 could not write CHAR array

ottlukas, 2022-11-09:
## Use case

Writing CHAR array to S7 PLc via plc4x sink kafka connector was throwing an error, as is shown in the code snippet:
```

"sinks.CommonPlcSetup.fields.steelGradeCurrent": "%DB851:136:CHAR[12]" 
```

## The error messages
```

[2022-03-01 03:22:16,673] INFO Configuring Bootstrap with Configuration{local-rack=1, local-slot=1,
remote-rack=0, remot-slot=5, pduSize=1024, maxAmqCaller=8, maxAmqCallee=8, controllerType='null'} (org.apache.plc4x.java.transport.tcp.TcpChannelFactory)
[2022-03-01
03:22:16,680] INFO S7 Driver running in ACTIVE mode. (org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic)
[2022-03-01
03:22:16,772] ERROR No DataTransportErrorCode for value 7 (org.apache.plc4x.java.s7.readwrite.DataTransportErrorCode)
[2022-03-01
03:22:18,340] ERROR No DataTransportErrorCode for value 7 (org.apache.plc4x.java.s7.readwrite.DataTransportErrorCode)
[2022-03-01
03:22:18,363] ERROR No DataTransportErrorCode for value 7 (org.apache.plc4x.java.s7.readwrite.DataTransportErrorCode)
[2022-03-01
03:22:18,363] ERROR No DataTransportErrorCode for value 7 (org.apache.plc4x.java.s7.readwrite.DataTransportErrorCode)

```

## Context

This was working with version 0.9.0-SNAPSHOT, but my customized version could no longer be compiled on 0.9.0-SNAPSHOT, so I had to compile the the code against 0.10.0-SNAPSHOT

 

Imported from Jira [PLC4X-338](https://issues.apache.org/jira/browse/PLC4X-338). Original Jira may contain additional context.
Reported by: hadoop.jz.

chrisdutz, 2024-05-17:
Ok ... I was able to confirm the issue and to resolve it in my setup ... could you please try the current SNAPSHOT version and verify if this issue is now fixed?

chrisdutz, 2024-05-17:
Ok ... also imported from Jira, so can't expect much ;-) ... I've got a test in my manual test that tests reading and writing of Char arrays and it works there.

closed 2024-05-17 by chrisdutz