apache/plc4x #644: Issue with Modbus Write -- PLC Connection Exception: Error Creating Channel

ottlukas, 2022-11-09:
Hi!
 
Thanks for supporting this library.
 
I'd like to seek your advice on an issue I'm having surrounding *writing* to Modbus registers.
 
I have a code snippet as follows:
 
!image-2022-05-30-13-31-56-320.png!
 
I'm able to make about ~100 requests successfully, until I encounter a`{{` PLC Connection Exception: Error creating channel error{}}}. This is consistent with multiple tries.
 
*I've tried using `{}CachedDriverManagers / PooledDriverManagers{`}, but this will result in a NullPointerException when I tried to use its `{}PlcConnection object to create a PlcWriteRequest.Builder{`}. Essentially, the "builder" object is null:*
 
!mb2.png!
 
*However, I'm able to create `{}PlcReadRequest builders for reading data; no NullPointerExceptions{`}:*
 
This is how I'm creating the `{}CachedDriverManager{`}:
 
!mb3.png!
 
I've noticed that I will never encounter a maximum buffer limit reached for connections created from the `{}CachedDriverManager{`}, used to make Modbus read requests. Hence, I considered using connections created from the CachedDriverManger to make Modbus write requests, __however I get a NullPointerException whenever I try to create a PlcWriteRequest.Builder object from CachedDriverManager__.
 
Hope you could help point me in the right direction, been stuck at this for a couple of days.
 
Thanks!
Regards,
Max
 

Imported from Jira [PLC4X-343](https://issues.apache.org/jira/browse/PLC4X-343). Original Jira may contain additional context.
Reported by: maxtan.

ottlukas, 2024-02-06:
@chrisdutz  After you tested a bunch of Modbus, is this still an open issue?

chrisdutz, 2024-02-06:
No longer an issue

closed 2024-02-06 by chrisdutz