apache/plc4x #648: roll back configurate dynamic properties

ottlukas, 2022-11-09:
Nifi integration has been pleasantly addressed in version 0.10.
Adding a record writer to the source processor is golden.

Unfortunately, the configuration has been changed, and the values to be read must be completed with dynamic properties.
In the previous version, there was a text field (PLC resource address String) where you could simply complete all the addresses to be read out.

Would it be possible to reverse this?  I have a project where I need to read out a lot of values via S7, adding this in NiFi now is endless work.

Imported from Jira [PLC4X-357](https://issues.apache.org/jira/browse/PLC4X-357). Original Jira may contain additional context.
Reported by: lintermans.

chrisdutz, 2023-09-22:
As the referenced ticket has been merged, can this issue be closed?

QuanticPony, 2023-09-25:
@chrisdutz It can be closed

closed 2023-09-25 by chrisdutz