apache/plc4x #649: The modbus rtu loop reads a large amount of data, causing a half packet problem

ottlukas, 2022-11-09:
```

2022-10-29 00:09:21.362  WARN   --- [ntLoopGroup-2-1] .p.j.s.GeneratedDriverByteToMessageCodec : Error
decoding package with content [0103c80fea000000000fe4000000000000000000000fdf00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000]:
Index 180 out of bounds for length 180java.lang.ArrayIndexOutOfBoundsException: Index 180 out of bounds
for length 180
    at com.github.jinahya.bit.io.ArrayByteInput.read(ArrayByteInput.java:106)
    at
com.github.jinahya.bit.io.DefaultBitInput.read(DefaultBitInput.java:57)
    at com.github.jinahya.bit.io.AbstractBitInput.unsigned8(AbstractBitInput.java:63)
 
  at com.github.jinahya.bit.io.AbstractBitInput.unsigned16(AbstractBitInput.java:92)
    at com.github.jinahya.bit.io.AbstractBitInput.readInt(AbstractBitInput.java:135)
 
  at com.github.jinahya.bit.io.AbstractBitInput.readInt(AbstractBitInput.java:118)
    at com.github.jinahya.bit.io.AbstractBitInput.readByte(AbstractBitInput.java:106)
 
  at org.apache.plc4x.java.spi.generation.ReadBufferByteBased.readSignedByte(ReadBufferByteBased.java:235)
 
  at org.apache.plc4x.java.spi.generation.ReadBuffer.readByte(ReadBuffer.java:44)
    at org.apache.plc4x.java.spi.generation.ReadBufferByteBased.readByteArray(ReadBufferByteBased.java:126)
 
  at org.apache.plc4x.java.modbus.readwrite.ModbusPDUReadHoldingRegistersResponse.staticParseBuilder(ModbusPDUReadHoldingRegistersResponse.java:112)
 
  at org.apache.plc4x.java.modbus.readwrite.ModbusPDU.staticParse(ModbusPDU.java:173)
    at org.apache.plc4x.java.modbus.readwrite.ModbusRtuADU.lambda$staticParseBuilder$0(ModbusRtuADU.java:135)
 
  at org.apache.plc4x.java.spi.codegen.io.DataReaderComplexDefault.read(DataReaderComplexDefault.java:70)
 
  at org.apache.plc4x.java.spi.codegen.io.DataReaderComplexDefault.read(DataReaderComplexDefault.java:61)
 
  at org.apache.plc4x.java.spi.codegen.fields.FieldReaderSimple.lambda$readSimpleField$0(FieldReaderSimple.java:34)
 
  at org.apache.plc4x.java.spi.codegen.FieldCommons.switchParseByteOrderIfNecessary(FieldCommons.java:69)
 
  at org.apache.plc4x.java.spi.codegen.fields.FieldReaderSimple.readSimpleField(FieldReaderSimple.java:34)
 
  at org.apache.plc4x.java.spi.codegen.fields.FieldReaderFactory.readSimpleField(FieldReaderFactory.java:129)
 
  at org.apache.plc4x.java.modbus.readwrite.ModbusRtuADU.staticParseBuilder(ModbusRtuADU.java:132)
 
  at org.apache.plc4x.java.modbus.readwrite.ModbusADU.staticParse(ModbusADU.java:121)
    at org.apache.plc4x.java.modbus.readwrite.ModbusADU.staticParse(ModbusADU.java:106)
 
  at org.apache.plc4x.java.modbus.rtu.ModbusRtuDriver.lambda$getStackConfigurer$0(ModbusRtuDriver.java:107)
 
  at org.apache.plc4x.java.spi.GeneratedDriverByteToMessageCodec.decode(GeneratedDriverByteToMessageCodec.java:87)
 
  at io.netty.handler.codec.ByteToMessageCodec$1.decode(ByteToMessageCodec.java:42)
    at io.netty.handler.codec.ByteToMessageDecoder.decodeRemovalReentryProtection(ByteToMessageDecoder.java:510)
 
  at io.netty.handler.codec.ByteToMessageDecoder.callDecode(ByteToMessageDecoder.java:449)
    at io.netty.handler.codec.ByteToMessageDecoder.channelRead(ByteToMessageDecoder.java:279)
 
  at io.netty.handler.codec.ByteToMessageCodec.channelRead(ByteToMessageCodec.java:103)
    at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
 
  at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
 
  at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:357)
 
  at io.netty.channel.DefaultChannelPipeline$HeadContext.channelRead(DefaultChannelPipeline.java:1410)
 
  at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
 
  at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
 
  at io.netty.channel.DefaultChannelPipeline.fireChannelRead(DefaultChannelPipeline.java:919)
    at
org.apache.plc4x.java.transport.serial.SerialChannel$SerialNioUnsafe.read(SerialChannel.java:289)
[... 62 lines cut ...]
org.apache.plc4x.java.transport.serial.SerialChannel$SerialNioUnsafe.read(SerialChannel.java:289)
 
  at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:722)
    at io.netty.channel.nio.NioEventLoop.processSelectedKeysPlain(NioEventLoop.java:623)
 
  at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:586)
    at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:496)
 
  at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:995)
 
  at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
    at io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
 
  at java.base/java.lang.Thread.run(Thread.java:833)
 
```

 

When I use modbus rtu to read 300 pieces of data, divided into four items for reading, half package will occur, causing the program to terminate

 

Imported from Jira [PLC4X-359](https://issues.apache.org/jira/browse/PLC4X-359). Original Jira may contain additional context.
Reported by: StrawberryBlue.

chrisdutz, 2024-02-10:
I doubt we can get in contact with the initial creator of this issue as JIRA is now closed. If I don't hear anything other, then I would close this issue when we're releasing the next version.

chrisdutz, 2024-02-10:
However the Serial Transport and the Modbus driver have recieved quite a bit of work in the last few days, would be cool to re-try it.

chrisdutz, 2024-06-30:
Ok ... if I don't get any feedback that suggests otherwise I'll close this issue the next time I go through the list.

closed 2024-08-02 by chrisdutz