apache/plc4x #653: The data returned from the slave station takes a bit-by-bit value.

x-em, 2022-11-14:
If the register is an input register or a hold register type, the data type is UINT (uint 16). In fact, the real project scenario represents values by specific bits, for example, each bit represents a switch state or alarm state. My idea is that this kind of data should be represented by a coil, but in fact, many project equipment take the actual value in this bitwise way, so I want to know if there is a better way to solve the problem besides forcing a method to do bit conversion to get the desired value.

example：
[image]

hutcheb, 2022-11-14:
Hi x-em,

Handling of Bit Strings is in a bit of state of flux as there is a push to get all the drivers returning exactly the same types.

However, for the Modbus driver you can use the format 400001:BOOL[16] to return a list of booleans.

Ben

chrisdutz, 2022-11-14:
But you should keep in mind that there's Big-Endian modbus and Little-Endian modbus ... so depending on what device you have, the order can differ.

x-em, 2022-11-15:
Thank you, ben. I will convert it separately on the program. Thank you, chrisdutz, for your hint.

closed 2022-11-15 by x-em