apache/plc4x #659: Kafka on-change/hysteresis support

dremsol, 2022-11-18:
Currently the Kafka Connect integration supports polling-based data acquisition. This results in duplicate messages when values don't change. Although some drivers support subscription not all devices seem to support it so a general implementation seems to be infeasible. What would be the best approach to on-change/hysteresis support in the Kafka integration?

hutcheb, 2022-11-18:
Hi Willem,

Each driver has the canRead, canWrite and canSubscribe attributes, these
can be used to decide between using a subscription based connection and a
read connection. I'm not 100% sure where to put this logic though,
somewhere within the Scraper?

Ben

chrisdutz, 2022-11-18:
We also have thought of implementing some "fake subscription" support based on polling, if a driver doesn't directly support subscriptions. However nobody ever volunteered to take on this action ... am I hearing a volunteer? ;-)

chrisdutz, 2024-08-16:
Are we really waiting for feedback here? Otherwise I think we might close this issue.

closed 2024-08-30 by chrisdutz