apache/plc4x #663: plc4j-Write a value to the coil, and the result is INVALID_ ADDRESS

isfangmk, 2022-11-21:
On version 0.10.0, when testing modbus protocol, write a value to the coil, and the result is INVALID_ ADDRESS. However, it is normal to test the read/write function of the holding register.

chrisdutz, 2022-11-21:
Well ... I have no Idea what type of address you used. From the release notes I can see that the modbus driver changed it's connection string. This is now "modbus-tcp" and no longer "modbus", as we've added "modbus-rtu" and "modbus-ascii" as additional flavors.

mkarier, 2023-08-14:
I have a similar problem but I am using 0.11.0-SNAPSHOT but I have noticed with wireshark that the address I put in actually one less then what I put in the code. So I am trying to write/read address 016385 and instead it is 16384. Once I added plus one it started to work. 
[image]

chrisdutz, 2023-08-14:
It is normal, that on the wire the numbers have an offset of one ... that's generally how the Modbus spec defines it ... Unfortunately I currently don't have the time to dig into this issue (if there is one), but I'll be happy to help you track it down, if you're willing to put in the effort. 

But I could also imagine, that this is one of the cases, where some PLC vendors don't implement the spec 100% correctly. Perhaps having a flag or switch in the connection string would be a solution for cases like this.

chrisdutz, 2023-09-22:
Can we close this issue?