apache/plc4x #664: S7-200 is not supported?

nmgwr, 2022-11-21:


nmgwr, 2022-11-21:
The Java version of the driver I use

chrisdutz, 2022-11-21:
Well we don't have anyone with such a device. Theoretically it should be possible to extend an S7 200 with an ethernet module, however I guess nobody has ever tested it with that. So in the classical sense: No it's not supported, but it might work. So please just give it a try.

heyoulin, 2022-11-22:
try my fork:  https://github.com/spnettec/plc4x

heyoulin, 2022-11-25:
Soory. I'm a bit busy recently, I will submit in few days

hongjinlin, 2022-12-26:
@nmgwr It's now supported at the develop branch, please try with the param "?controller-type=S7_200”. 
Welcome to feedback if you still have any problems.

nmgwr, 2022-12-28:
@chrisdutz In some industrial control systems in China, a large number of low-end PLCs use the s7-200. I think the author's development branch supports it. Let me try

nmgwr, 2022-12-28:
@heyoulin thank you

nmgwr, 2022-12-28:
@heyoulin thank you，i try， thank you

chrisdutz, 2023-09-22:
Can we close this issue?

chrisdutz, 2024-02-06:
Closing issue as S7-1200 is supported.

closed 2024-02-06 by chrisdutz