apache/plc4x #667: "Unable to Close 'broken' Connection" when extracting data from Rockwell PLC 1756-EN2T

johanngil, 2022-11-23:
Dear community, 
With a recent (from Friday last week) build from PLC4X 0.11.0-SNAPSHOT (https://github.com/apache/plc4x/tree/logix_develop) we connected our Kafka environment through the PLC4X connector to our Rockwell PLC of type 1756-EN2T to extract data from the PLC. We used the following properties settings:
{
"connector.class": "org.apache.plc4x.kafka.Plc4xSourceConnector",
"tasks.max": "1",
  "default.topic": "DefaultTopic",
"sources": "MHSLineTakeway",
"sources.MHSLineTakeway.connectionString": "opcua:tcp://143.21.40.239:44818",
        "sources.MHSLineTakeway.pollReturnInterval": "5000",
        "sources.MHSLineTakeway.bufferSize": "1000",
        "sources.MHSLineTakeway.jobReferences": "data-acquisition",
        "sources.MHSLineTakeway.jobReferences.data-acquisition.topic": "data-acquisition-topic",
        "jobs": "data-acquisition",
        "jobs.data-acquisition.interval": "100",
        "jobs.data-acquisition.fields": "stsmessageid",
        "jobs.data-acquisition.fields.stsmessageid": "PRIME_SSB_IO/STS_TRIGGERID",
        "bootstrap.servers": "localhost:9092",
        "jobs.data-acquisition.tags": "stsmessageid",
        "jobs.data-acquisition.tags.stsmessageid": "PRIME_SSB_IO/STS_TRIGGERID:INT"
}

These resulted in the below error (attached the entire log file):
C[2022-11-23 08:21:50,517] DEBUG [mhs-linetakeway-source|task-0] Connection was detected as broken and is invalidated in Cached Manager (org.apache.plc4x.java.utils.connectionpool2.CachedDriverManager:127)
[2022-11-23 08:21:50,518] WARN [mhs-linetakeway-source|task-0] Broken Connection was returned, although it is not borrowed, currently. (org.apache.plc4x.java.utils.connectionpool2.CachedDriverManager:132)
[2022-11-23 08:21:50,518] DEBUG [mhs-linetakeway-source|task-0] Unable to Close 'broken' Connection (org.apache.plc4x.java.utils.connectionpool2.CachedDriverManager:138)
java.lang.NullPointerException

Do you know what the root cause of the problem is? Thank you for any assistance!
[trace_connect_error.txt](https://github.com/apache/plc4x/files/10074752/trace_connect_error.txt)

hutcheb, 2022-11-23:
Hi johanngil,

That's awesome that you built the logix_develop branch :) It looks as though you are trying to use the opcua driver to communicate with it.

The connection string will be something like "logix:tcp://143.21.40.239:44818?backplane=1&slot=1"

I haven't spent a lot of time on the reading UDT's but the general format of the address will be PRIME_SSB_IO.STS_TRIGGERID:INT

Ben

hutcheb, 2022-11-23:
Sorry, also just realized I gave you bad information, it is the eip_update branch that is more up to date.

https://github.com/apache/plc4x/pull/405

chrisdutz, 2023-09-22:
Is this issue still valid (A lot has been updated in the EIP driver)

DamonDBT, 2023-10-07:
i use 0.10.0 version by java to connect s7-400, most time it works well ,but recently got the error , i think this error maybe have some relation to this issue:

14:00:56.477 [ForkJoinPool.commonPool-worker-11] WARN  o.a.p.j.u.c.CachedDriverManager - [lambda$getConnection_$0,229] - **Unable to establish connection to PLC** s7://176.16.11.185?remote-rack=0&remote-slot=3
org.apache.plc4x.java.api.exceptions.PlcConnectionException: Error creating channel.
	at org.apache.plc4x.java.spi.connection.NettyChannelFactory.createChannel(NettyChannelFactory.java:143)
	at org.apache.plc4x.java.spi.connection.DefaultNettyPlcConnection.connect(DefaultNettyPlcConnection.java:123)
	at org.apache.plc4x.java.PlcDriverManager.getConnection(PlcDriverManager.java:78)
	at com.ruoyi.level2.config.CommonConfig.lambda$null$0(CommonConfig.java:84)
	at org.apache.plc4x.java.utils.connectionpool2.CachedDriverManager.lambda$getConnection_$0(CachedDriverManager.java:219)
	at java.util.concurrent.CompletableFuture$AsyncRun.run(Unknown Source)
	at java.util.concurrent.CompletableFuture$AsyncRun.exec(Unknown Source)
	at java.util.concurrent.ForkJoinTask.doExec(Unknown Source)
	at java.util.concurrent.ForkJoinPool$WorkQueue.runTask(Unknown Source)
	at java.util.concurrent.ForkJoinPool.runWorker(Unknown Source)
	at java.util.concurrent.ForkJoinWorkerThread.run(Unknown Source)
Caused by: io.netty.channel.AbstractChannel$AnnotatedNoRouteToHostException: **No route to host: no further information:** /176.16.11.185:102
Caused by: java.net.NoRouteToHostException: No route to host: no further information
	at sun.nio.ch.SocketChannelImpl.checkConnect(Native Method)
	at sun.nio.ch.SocketChannelImpl.finishConnect(Unknown Source)
	at io.netty.channel.socket.nio.NioSocketChannel.doFinishConnect(NioSocketChannel.java:330)
	at io.netty.channel.nio.AbstractNioChannel$AbstractNioUnsafe.finishConnect(AbstractNioChannel.java:334)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:707)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:655)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:581)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:493)
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:989)
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	at io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
	at java.lang.Thread.run(Unknown Source)
14:00:57.096 [RuoyiScheduler_Worker-12] WARN  c.r.l.t.TrackingStrand - [Track,511] - S1  castSpeed:  0.96
14:00:57.473 [RuoyiScheduler_Worker-1] WARN  o.a.p.j.u.c.CachedDriverManager - [handleBrokenConnection,132] - Broken Connection was returned, although it is not borrowed, currently.
14:00:57.473 [RuoyiScheduler_Worker-1] WARN  c.r.l.t.WrappingPlc - [doWrite,575] - --------Connection exception in trying to connect
org.apache.plc4x.java.api.exceptions.PlcConnectionException: No Connection Available, timed out while waiting in queue.
	at org.apache.plc4x.java.utils.connectionpool2.CachedDriverManager.getConnection(CachedDriverManager.java:182)
	at com.ruoyi.level2.task.WrappingPlc_doWriteWibu_10.doWrite(Unknown Source)
	at com.ruoyi.level2.task.WrappingPlc.doWrite(WrappingPlc.java)
	at com.ruoyi.level2.task.WrappingPlc.sp_doWrite_114(WrappingPlc.java)
	at com.ruoyi.level2.task.WrappingPlc_writingToPlcWibu_3.writingToPlc(Unknown Source)
	at com.ruoyi.level2.task.WrappingPlc.writingToPlc(WrappingPlc.java)
	at com.ruoyi.level2.task.WrappingPlc.sp_writingToPlc_108(WrappingPlc.java)
	at com.ruoyi.level2.task.WrappingPlc_WrapWibu_1.Wrap(Unknown Source)
	at com.ruoyi.level2.task.WrappingPlc.Wrap(WrappingPlc.java)
	at sun.reflect.GeneratedMethodAccessor684.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at com.ruoyi.quartz.util.JobInvokeUtil.invokeMethod(JobInvokeUtil.java:61)
	at com.ruoyi.quartz.util.JobInvokeUtil.invokeMethod(JobInvokeUtil.java:33)
	at com.ruoyi.quartz.util.QuartzDisallowConcurrentExecution.doExecute(QuartzDisallowConcurrentExecution.java:19)
	at com.ruoyi.quartz.util.AbstractQuartzJob.execute(AbstractQuartzJob.java:46)
	at org.quartz.core.JobRunShell.run(JobRunShell.java:202)
	at org.quartz.simpl.SimpleThreadPool$WorkerThread.run(SimpleThreadPool.java:573)
Caused by: java.util.concurrent.TimeoutException: null
	at java.util.concurrent.CompletableFuture.timedGet(Unknown Source)
	at java.util.concurrent.CompletableFuture.get(Unknown Source)
	at org.apache.plc4x.java.utils.connectionpool2.CachedDriverManager.getConnection(CachedDriverManager.java:179)
	... 17 common frames omitted
14:00:57.489 [ForkJoinPool.commonPool-worker-11] WARN  o.a.p.j.u.c.CachedDriverManager - [lambda$getConnection_$0,229] - Unable to establish connection to PLC s7://176.16.11.180?remote-rack=0&remote-slot=3
org.apache.plc4x.java.api.exceptions.PlcConnectionException: Error creating channel.
	at org.apache.plc4x.java.spi.connection.NettyChannelFactory.createChannel(NettyChannelFactory.java:143)
	at org.apache.plc4x.java.spi.connection.DefaultNettyPlcConnection.connect(DefaultNettyPlcConnection.java:123)
	at org.apache.plc4x.java.PlcDriverManager.getConnection(PlcDriverManager.java:78)
	at com.ruoyi.level2.config.CommonConfig.lambda$null$0(CommonConfig.java:84)
[... cut ...]
	at com.ruoyi.quartz.util.AbstractQuartzJob.execute(AbstractQuartzJob.java:46)
	at org.quartz.core.JobRunShell.run(JobRunShell.java:202)
	at org.quartz.simpl.SimpleThreadPool$WorkerThread.run(SimpleThreadPool.java:573)
Caused by: java.util.concurrent.TimeoutException: null
	at java.util.concurrent.CompletableFuture.timedGet(Unknown Source)
	at java.util.concurrent.CompletableFuture.get(Unknown Source)
	at org.apache.plc4x.java.utils.connectionpool2.CachedDriverManager.getConnection(CachedDriverManager.java:179)
	... 13 common frames omitted

chrisdutz, 2023-10-07:
Did you try with the version 0.11.0 that we released just yesterday?

DamonDBT, 2023-10-08:
i have not try the 0.11.0 yet,  i will try in some days later when i go to the factory, if i have some  result, i will report.

chrisdutz, 2024-05-17:
What's the status of this issue?

chrisdutz, 2024-06-30:
Can this issue be closed?

johanngil, 2024-07-01:
Business Use

Yes, please close.

Regards,
Dietmar Giljohann

closed 2024-07-01 by chrisdutz