apache/plc4x #669: Ensure the KNX date and time fields are defined correctly

chrisdutz, 2022-11-24:
In KNX a Time needs to be mapped to a DATE_AND_TIME type according to 03_07_02 Datapoint Types v01.08.02 AS.pdf
and in genereal I think we should really review our MSPEC and XSLTs for KNX that they generate the correct DataIO for us.

splatch, 2024-07-05:
@chrisdutz given that you worked a lot on KNX stuff, do you think this issue is still valid?

chrisdutz, 2024-07-05:
The thing is that the Go version of the KNX driver is way more advanced than the Java version ... however I see no way in me being able to port that back any time soon. But I think we can indeed close this issue, as I should probably open a new one about porting back the KNX advancements.

closed 2024-07-05 by chrisdutz