apache/plc4x #694: Implement custom exception handling in TriggeredScraperTask

frido, 2022-12-14:
If there is exception while scraping, the exception is logged on the DEBUG level.
https://github.com/apache/plc4x/blob/d2e3dcafee07b9406489bd02c84b1f8fe432601f/plc4j/tools/scraper/src/main/java/org/apache/plc4x/java/scraper/triggeredscraper/TriggeredScraperTask.java#L204

The user can override this `handleException(Exception e)` method, but it is not possible to pass custom implementation of `TriggeredScraperTask` to the `TriggeredScraperImpl`.
There is hard-coded `TriggeredScraperTask` class:
https://github.com/apache/plc4x/blob/d2e3dcafee07b9406489bd02c84b1f8fe432601f/plc4j/tools/scraper/src/main/java/org/apache/plc4x/java/scraper/triggeredscraper/TriggeredScraperImpl.java#L202

It should be possible to set something like _task factory_ to the `TriggeredScraperImpl`.

chrisdutz, 2022-12-20:
Yeah ... unfortunately the scraper is sort of a part we are planning on rewriting as the original author is no longer seriously active here and we're not too happy with parts of it. Would this be something you would be willing to help with? I know we just recently had someone step up raising his hand for starting to clean up this part. Perhaps something you could join in on?

chrisdutz, 2024-08-16:
Hmmm ... are we really waiting for feedback here? If not, I think we can close it.

closed 2024-08-30 by chrisdutz