apache/plc4x #695: Invalid url for TCP transport

floka94, 2022-12-15:
I try to connect to a S7 PLC on Slot 3. I tried several connection strings:

```java
"s7://172.16.0.12/0/3"
"s7://172.16.0.12/0/3?controller-type=S7_400"
"s7://172.16.0.12/3/0?controller-type=S7_400"
"s7://172.16.0.12/3/0"
```
I always get an error:

```bash
Caused by: org.apache.plc4x.java.api.exceptions.PlcRuntimeException: Invalid url for TCP transport
	at org.apache.plc4x.java.transport.tcp.TcpTransport.createChannelFactory(TcpTransport.java:57)
	at org.apache.plc4x.java.spi.connection.GeneratedDriverBase.getConnection(GeneratedDriverBase.java:148)
	at org.apache.plc4x.java.spi.connection.GeneratedDriverBase.getConnection(GeneratedDriverBase.java:99)
	at org.apache.plc4x.java.PlcDriverManager.getConnection(PlcDriverManager.java:77)
	at org.apache.plc4x.java.opm.PlcEntityInterceptor.refetchAllFields(PlcEntityInterceptor.java:218)
	... 16 more
```

chrisdutz, 2022-12-15:
Which version of PLC4X are you using? I think we've changed that rack and slot setting to use a default and to be overridable via query options ... please have a look here: https://plc4x.apache.org/users/protocols/s7.html

So I would assume you have to use something like:
`
"s7://172.16.0.12"
"s7://172.16.0.12?remote-rack=1&remote-slot=2"
"s7://172.16.0.12?remote-rack=1&remote-slot=2&controller-type=S7_400"
`
If you want to override the local rack and slot, there are counterparts for that.

floka94, 2022-12-15:
I use version 0.10.0.

Ah thank you very much! It seems I used an outdated code example! Now everything is working as expected!

closed 2022-12-15 by floka94