apache/plc4x #701: Cannot write singe BOOL or BYTE to PLC with Object PLC Mapping

floka94, 2022-12-16:
I am trying to write a single BOOL value to a Siemens S7-400 PLC with the OPM Feature.

Here is my Entity:

```java
@PlcEntity
public class NewDataSetReady {

    @PlcField("%DB9002:5.0:BOOL")
    public boolean isDataSetReady;

    public NewDataSetReady() {
        // For OPC
    }

}
```

Here is my write operation:

```java
NewDataSetReady newDataSetReady = new NewDataSetReady();
newDataSetReady .setIsDataSetReady(false);
NewDataSetReady newDataSetReadyResponse = entityManager.write(NewDataSetReady.class, connectionString, newDataSetReady);
```

After the write operation i get the following warning and my value is not persisted:
```bash
[org.apa.plc.jav.s7.rea.pro.S7ProtocolLogic] (nioEventLoopGroup-2-1) Got an error response from the PLC. This particular response code usually indicates that PUT/GET is not enabled on the PLC.
```

Here is a snapshot from the ISOTCP Protocoll communication between PLC4X and S7-400 (from Wireshark):

Write data to PLC:
[image]

Response from PLC:
[image]

I have also tried the whole thing with a single byte, but get the same error. For other datatypes like INT the write process works. 

I tried the BIT write operation also on a S7-1500 and it worked.

---
PLC4X v. 0.10.0

chrisdutz, 2022-12-20:
Admittedly the person who wrote this part of PLC4J is currently not really active anymore ... would this be something you would be willing to look into?

floka94, 2023-01-25:
I thought that all this features will work. It seems that there are more bugs and problems with this library and is not ready for production usage :-(

splatch, 2023-01-25:
@floka94 The S7 protocol is reverse engineered hence its stability can vary depending on many things. The OPM is just an utility which was invented more than three years ago and haven't received much of maintenance since then. If you run into troubles with protocols implemented by plc4x please raise issues with appropriate input.

In order to diagnose issues we need a PCAP capture *or* working test case with sample request and PLC response, if it generates one. You can copy packet contents into XML test suites we have for protocols (see [serializer test for S7)](https://github.com/apache/plc4x/blob/develop/protocols/s7/src/test/resources/protocols/s7/ParserSerializerTestsuite.xml) to confirm frame is parsed and mapped properly.

Your screenshots are helpful but they can't help with bug fixes as correlation between them and problem is not possible.

glcj, 2023-01-26:
Hi,

I was able to see the problem, tested with a S7-400.
I think it may be the "Return code"  in the "Item" section, currently 0x77 and should be 0x00.
I inform you,

Best regards,

floka94, 2023-01-28:
I find the OPM tool very useful and would be happy to use it in our productive environment once the bugs are fixed. Are there plans to further develop or maintain the opm tool in the future?

splatch, 2023-01-28:
@floka94 its open source project, so as for pther open source tools motto is "bring your own effort". Since original maintainer lost interest in keeping it up, nothing blocks you from stepping in and fixing it.

DamonDBT, 2023-07-27:
Is it true that PLC4X（version 0.10.0 called by java） cannot write boolean values to a S7-400 PLC? @chrisdutz 
 I encountered the same issue. 
With Kepware, it is possible to write to the same DB address.
I would like to confirm this functionality to avoid attempting to write boolean variables to the 400 PLC in the future.
thanks.

chrisdutz, 2023-07-27:
Well it should be able to do so ... if it doesn't work, that's probably a bug someone should report and open a new issue for.

And then all it needs is someone willing to fix it ;-)

chrisdutz, 2023-09-22:
I extended the OPM example by quite a bit and added some (commented) out line to also write back the values ... I couldn't reproduce the error you were reporting. So could you please try it again with the latest SNAPSHOT and confirm if it's resolved or still present?

chrisdutz, 2024-02-10:
So ... I have tested the PLC4X functionality for writing values to boolean fields on 2 S7-1200 devices and one S7-1500 ... I can only say that it works on these devices. Unfortunately I don't have an S7-300 or S7-400, but @glcj ... can you provide any input on this matter? Is the current version still not able to write BOOL values to an S7-400?

chrisdutz, 2024-08-16:
Closing for lack of activity.

closed 2024-08-16 by chrisdutz