apache/plc4x #702: Cannot read REAL value from S7-1500 with PLC4J

floka94, 2022-12-19:
I cannot read REAL values from a S7-1500 with PLC4J with version 0.10.0.

I get the following error from the PLC4J Lib:

```bash
Caused by: java.lang.NullPointerException: Cannot invoke "org.apache.plc4x.java.s7.readwrite.S7PayloadReadVarResponse.getItems()" because "payload" is null
```

Screenshots from Wireshark:

Request from PLC4J to read REAL:
[image]

Response from S7-1500:
[image]

chrisdutz, 2022-12-20:
The odd thing is that I have a test with an S7-1200 to test all data-types ... this is a regular data-type which my PLC supports and I couldn't see any problems. 

Also that wireshark can't decode the response sort of seems to indicate that the problem might be your PLC. Is this a genuine Siemens S7-1500 or a clone?

floka94, 2023-01-26:
@chrisdutz this was a simulated S7 with the S7-PLCSIM Advanced V3.0.

glcj, 2023-01-26:
Hi,

Driver version: 0.11.0-SNAPSHOT

For CPU S7-300, work! (%MD100:REAL)
For CPU S7-400, work!
For CPU S7-1214C, work!
For CPU S7-1500,  S7-PLCSIM Advanced V4.0. don't work!

We will have to perform other tests to find out the difference,

glcj, 2023-01-27:
Hi  floka94,

You can check if you have enabled PUT/GET reading on the CPU, you should do that before transferring the project.

If so, please indicate so that the work team can close this point.

For CPU S7-300, work! (%MD100:ACTUAL)
For CPU S7-400, work!
For CPU S7-1214C, work!
For CPU S7-1500, S7-PLCSIM Advanced V4.0. work!

Best regards,

floka94, 2023-01-27:
I updated my code and dependencies

```xml
        <dependency>
            <groupId>org.apache.plc4x</groupId>
            <artifactId>plc4j-api</artifactId>
            <version>0.11.0-SNAPSHOT</version>
        </dependency>
        <dependency>
            <groupId>org.apache.plc4x</groupId>
            <artifactId>plc4j-driver-s7</artifactId>
            <version>0.11.0-SNAPSHOT</version>
            <scope>runtime</scope>
        </dependency>
```

It works for other datatypes like BOOL but still not for REAL using S7-PLCSIM Advanced V3.0.

glcj, 2023-01-27:
Hello,

I already tested and found no problems:

1. Verify that the DB is not optimized.
2. Verify that the PLC does not have security enabled.
3. Verify that the CPU supports PUT/GET.

I'm preparing a full test run for the S7 driver. I uploaded a simple reading example in [1].

I hope it is useful for you.

1. https://github.com/glcj/PLC4XS7ExamplesV9/blob/0.11.0/src/main/java/com/ceos/plc4x/s7/PLCReadTags.java

floka94, 2023-01-27:
As I already said, it works for any other datatype but not for REAL. So the problem can not be because of the three points you mentioned.

Thank you for the demo project. I will give it a try and inform you.

floka94, 2023-01-28:
Unfortunately the same Exception with your demo project.

```bash
[main] INFO org.apache.plc4x.java.DefaultPlcDriverManager - Instantiating new PLC Driver Manager with class loader jdk.internal.loader.ClassLoaders$AppClassLoader@4b1210ee
[main] INFO org.apache.plc4x.java.DefaultPlcDriverManager - Registering available drivers...
[main] INFO org.apache.plc4x.java.DefaultPlcDriverManager - Registering driver for Protocol s7 (Siemens S7 (Basic))
[main] INFO org.apache.plc4x.java.transport.tcp.TcpChannelFactory - Configuring Bootstrap with Configuration{local-rack=1, local-slot=1, local-tsap=0, remote-rack=0, remote-slot=2, remote-tsap=0, pduSize=1024, maxAmqCaller=8, maxAmqCallee=8, controllerType='null'}
[nioEventLoopGroup-2-1] INFO org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic - S7 Driver running in ACTIVE mode.
[main] INFO org.apache.plc4x.java.spi.connection.NettyChannelFactory - Channel is closed, closing worker Group also
[main] INFO org.apache.plc4x.java.spi.connection.NettyChannelFactory - Worker Group was closed successfully!
java.util.concurrent.ExecutionException: java.lang.NullPointerException: Cannot invoke "org.apache.plc4x.java.s7.readwrite.S7PayloadReadVarResponse.getItems()" because "payload" is null
	at java.base/java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:396)
	at java.base/java.util.concurrent.CompletableFuture.get(CompletableFuture.java:2096)
	at com.ceos.plc4x.s7.PLCReadTags.main(PLCReadTags.java:32)
Caused by: java.lang.NullPointerException: Cannot invoke "org.apache.plc4x.java.s7.readwrite.S7PayloadReadVarResponse.getItems()" because "payload" is null
	at org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.decodeReadResponse(S7ProtocolLogic.java:765)
	at org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.lambda$16(S7ProtocolLogic.java:231)
	at java.base/java.util.concurrent.CompletableFuture$UniApply.tryFire(CompletableFuture.java:646)
	at java.base/java.util.concurrent.CompletableFuture.postComplete(CompletableFuture.java:510)
	at java.base/java.util.concurrent.CompletableFuture.complete(CompletableFuture.java:2147)
	at org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.lambda$23(S7ProtocolLogic.java:269)
	at org.apache.plc4x.java.spi.Plc4xNettyWrapper.decode(Plc4xNettyWrapper.java:187)
	at io.netty.handler.codec.MessageToMessageCodec$2.decode(MessageToMessageCodec.java:81)
	at io.netty.handler.codec.MessageToMessageDecoder.channelRead(MessageToMessageDecoder.java:88)
	at io.netty.handler.codec.MessageToMessageCodec.channelRead(MessageToMessageCodec.java:111)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:442)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:412)
	at io.netty.handler.codec.ByteToMessageDecoder.fireChannelRead(ByteToMessageDecoder.java:346)
	at io.netty.handler.codec.ByteToMessageDecoder.channelRead(ByteToMessageDecoder.java:318)
	at io.netty.handler.codec.ByteToMessageCodec.channelRead(ByteToMessageCodec.java:103)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:442)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:412)
	at io.netty.channel.DefaultChannelPipeline$HeadContext.channelRead(DefaultChannelPipeline.java:1410)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:440)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
	at io.netty.channel.DefaultChannelPipeline.fireChannelRead(DefaultChannelPipeline.java:919)
	at io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:166)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:788)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:724)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:650)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:562)
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:997)
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	at io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
	at java.base/java.lang.Thread.run(Thread.java:833)
EventDispacher Bye!
ObjectProcessor Bye!
```

Here is ma recorded pcapng file, I hobe this can help you to find out the problem:
[s7_real.zip](https://github.com/apache/plc4x/files/10528144/s7_real.zip)

glcj, 2023-01-28:
Hello,

In the captures (pcap) you can clearly see that the problem is in the PLCSIM response, specifically the length of the "Data.Item" is 32, when it should be 4 (the length of a real in bytes).

The request job is correct byte for byte, relative to my local test.

The value of 32 would be valid if the "Transport size" were in bits.

For the moment I must point out that the problem is in the PLCSIM.

I did the test with version 4.0 of PLCSIM, please, if you can download and try with this version, you have a 15-day trial license.

I will be careful,
Best regards,

heyoulin, 2023-02-13:
I have tested it. Try  [https://github.com/spnettec/plc4x.git](url) .  heyoun branch

chrisdutz, 2023-09-22:
I did quite a number of updates in the Java S7 driver. Could you please give the current SNAPSHOT another try? However I did test the writing of REAL values with my S7 1200 and that worked flawlessly.

chrisdutz, 2024-02-06:
Should be supported now.

closed 2024-02-06 by chrisdutz

NdiayeeCheikh, 2025-05-08:
Hi you'll
I'm having this issues , what does it mean :
[com.example.plcapp.ModbusClient.main()] INFO org.apache.plc4x.java.DefaultPlcDriverManager - Instantiating new PLC Driver Manager with class loader org.codehaus.mojo.exec.URLClassLoaderBuilder$ExecJavaClassLoader@323a47fe
[com.example.plcapp.ModbusClient.main()] INFO org.apache.plc4x.java.DefaultPlcDriverManager - Registering available drivers...
[com.example.plcapp.ModbusClient.main()] INFO org.apache.plc4x.java.DefaultPlcDriverManager - Registering driver for Protocol modbus-ascii (Modbus ASCII)
[com.example.plcapp.ModbusClient.main()] INFO org.apache.plc4x.java.DefaultPlcDriverManager - Registering driver for Protocol modbus-rtu (Modbus RTU)
[com.example.plcapp.ModbusClient.main()] INFO org.apache.plc4x.java.DefaultPlcDriverManager - Registering driver for Protocol modbus-tcp (Modbus TCP)
[com.example.plcapp.ModbusClient.main()] INFO org.apache.plc4x.java.transport.serial.SerialChannelFactory - Configuring Bootstrap with org.apache.plc4x.java.transport.serial.DefaultSerialTransportConfiguration@6cac4b00
^C/media/sda4/plc4x/plc-demo #