apache/plc4x #703: [ADS] The encoding used for strings is incorrect

inosik, 2022-12-19:
I noticed that PLC4X uses UTF-8 encoding to read/write strings to ADS devices. According to [Beckhoffs documentation](https://infosys.beckhoff.com/content/1033/tc3_plc_intro/2529327243.html?id=4470464088993876920), the correct encoding for `STRING` variables is Windows-1252. I created a demo program in TwinCAT 3, with a string value being set to the Euro-sign:

[image]

[image]

_Note: `BinaryString8` is a union type, consisting of a string and a byte array, which allows me to see the binary data representing the string. For comparison, there's also a version with a `WSTRING`, which is encoded using UTF-16._

As you can see, the Euro-sign is encoded as 128 (`0x80`), while [in UTF-8 it should be `0xe282ac`](https://www.fileformat.info/info/unicode/char/20ac/index.htm).

chrisdutz, 2022-12-20:
Should be pretty simple to change ... thanks for reporting this. Would you be willing to provide a Pull-Request with a fix?

inosik, 2022-12-20:
Sure, I‘d be happy to fix this!

chrisdutz, 2022-12-20:
I guess you should probably have a look at the ads.mspec document in /protocols/ads ... there find the DataItem definition and update the encoding of the types.

However you probably need to add support for the encoding to the org.apache.plc4x.java.spi.generation.ReadBufferByteBased (readString) and it's counterpart org.apache.plc4x.java.spi.generation.WriteBufferByteBased (writeString) ... if you want (don't have to) ... we've got the same concept for PLC4C (plc4c/spi/src/read_buffer.c and plc4c/spi/src/write_buffer.c) and PLC4Go (plc4go/spi/utils/ReadBufferByteBased.go and plc4go/spi/utils/WriteBufferByteBased.go)

chrisdutz, 2023-09-22:
So is there any progress on this issue?

inosik, 2023-09-25:
Sorry for the silence, I didn't come around setting up a Java dev environment yet. But I see that you recently fixed this, so I guess this issue can be closed 👍

closed 2023-09-25 by inosik