apache/plc4x #714: [Feature Request]: 建议(Suggestion)参考 NI OPC Server (National Instruments OPC Server) 与 Labview 的交互方式(Shared Variables)

chzhm159, 2022-12-28:
### What would you like to happen?

软件架构可以参考 NI OPC Server (National Instruments OPC Server) 与 Labview 的交互方式(Shared Variables). 
 (The software architecture can refer to the interaction between NI OPC Server (National Instruments OPC Server) and Labview.(Shared Variables))

1. --- --------------- TCP/UDP/Serial Port... Connectiong Manager --------------
  1.1 IP,Port,Device Type etc.                                          
  1.2 Disconnection Retry etc.                                          
  1.3 Connection Event Emit etc.             
                           
2. ------------------  Protocol Adapter --------------------------------------------
  2.1 Siemens/Omron/... PLC Communication Protocol Adapter  
  2.2 Read/Write
  2.3 Data Type Convert

3. ------------------ Data Cache(Memory Cache etc.) Layer----------------------
  3.1 Variable Binding (Variable Mapping?)                                  
  3.2 Data Refresh(Active, Passive, Timed etc.)

4. ------------------ Application Layer ---------------------------------------------
  The application layer will be very simple based on the data 
  cache layer. In the application layer, you only need to 
  declare variables, get values, and assign values.

### Programming Languages

- [ ] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

hutcheb, 2023-01-01:
Hi chzhm159,

The NI OPC Server is primarily an OPC-DA server. There is no OPC-DA driver within PLC4X. 

However, I believe there is an optional package for the NI OPC server for it to support OPC-UA. You could then use the OPC-UA driver in PLC4X.

In either case, If you could outline a bit more about what you are trying to achieve and your architecture it would help.

Ben

chzhm159, 2023-01-06:
使用 Java 举例,PLC 使用 KEYENCE KV-7500,读取一个32位有符号整数.

Using Java as an example, PLC uses KEYENCE KV-7500 to read a 32-bit signed integer.

First step:
  configuration variables(Device information and register information)
```
{
  "devices": [                      
    {
      "deviceID": "L1:D1",          // device unique identifier(设备全局唯一编号)
      "deviceName": "station-1",
      "deviceModel": "kv-7500",     
      "protocolType": "upperlink",  
      "host": "192.168.100.1",      
      "port": 8501,
      "connectTimeout": 3000,       
      "retryInterval": 3000,        
      "tags": [
        {
          "tagName": "output",
          "key": "L1:D1:output",    // Globally unique key for each tag(Tag全局唯一编号)
          "registerType": "FM",     // register area (plc寄存器区域名称)
          "registerIndex": 100,     // register index (寄存器区编号)
          "offset":0,
          "unit": "int32",          // register data type (寄存器数据类型)
          "count": 3,
          "readInterval": 10,       // Read cycle, in milliseconds (读取周期,单位毫秒)
          "timeout": 60000,                 
          "operate": "r",           // read or write or (read and write) (读/写/(读写))
        }
      ]
    }
  ]
}
```
Framework:
  框架的工作就是解析配置,建立与plc的链接,同时具备失败重连的特性,同时根据每个Tag的配置自动
  读取对于的数据缓存起来.将链接管理与协议解析对编程人员完全屏蔽,仅需要了解plc基本的寄存器
  区域,名称,数据类型等通用概念即可
  The job of the framework is to parse the configuration, link the device, and automatically collect the data of the PLC and save it to the cache

Second step:
  业务代码就可以非常简单,完全不关心plc的链接与通信协议
  My business code can be written like this:
```
public class Demo{

  public void onConnectSuccessfully(){
    // connection succeeded (链接成功)
  }
  
  public void onConnectFailed(){
   // connection failed (链接失败)
  }

  public void onDisconnected(){
   // lost connection (连接中断开)
  }

  public void onClosed(){
   // connection closed (连接关闭)
  }
  /**
  * Called according to functional needs (根据功能需要来调用)
  */  
  public void onOutput(){
    // Get instance by unique identifier for each variable
    Tag otherTag = TagMangager.getTag("L1:D1:output");

    // ==== read value ===    
    
    // Read mode 1. Read the original value, which can default to Byte
    Byte[] raw = tag.getRawData();
    int raw = DataUtils.convert2Int(raw);
    
    // Read mode 2. Directly obtain the value of a specific type
    int raw = tag.getInt();

    // >>>>> Business logic....save data to database, calculate, judge, etc.

    // ==== [Optional] write value ===
    tag.write(Byte[] data);

    // ==== [Optional] Read and write values to other variables ===
    Tag otherTag = TagMangager.getTag("key");
    otherTag.getXXX();
    otherTag.write(Data);
  }
}
```

hutcheb, 2023-01-25:
Hi chzhm159,

Sorry I am still a little confused about what exactly you're requesting. The example you've given seems to use some sort of Modbus derivative, yet the original was a request to support an NI OPC Server.

Can you please provide some more details about what you're trying to communicate to, or if you have a packet capture from from another framework you'd like to see in PLC4X.

Kind Regards

Ben

chzhm159, 2023-01-30:
首先,为我给您带来的困惑表示歉意
First of all, apologies for the confusion I caused you.
我希望能做出一些贡献,当我对plc4x学习更多的之后
I hope to make some contributions as I learn more about PLC4X

closed 2023-01-30 by chzhm159