apache/plc4x #716: [Feature Request]: Apache Camel Consumer repeated polling

SteinOv, 2022-12-28:
### What would you like to happen?

It would be useful if the Plc4XConsumer would be able to poll the state of tags every x seconds/milliseconds. There is the possibility of adding a "trigger" with a period (I believe in milliseconds). I couldn't figure out exactly how this works from the documentation, but it seems that a plc tag needs to trigger the polling instead of it being solely time-based.
Right now I'm using a work-around in my own code by using a PollingConsumer that stops and starts the Plc4XConsumer for every request. But this isn't ideal.
So, I wanted to add the possibility to poll plc tags every x seconds/milliseconds.

I have two possibilities in mind for configuring this:
- The polling is activated if a period is configured without a trigger
- Add a new field for activating the polling, I could call it "pollingInterval" for example

I would appreciate some input for what is preferred.

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2023-01-03:
I do think that in general it should be polling and subscription based triggers are an option ... becaust initially almost no protocol supported subscriptions.

SteinOv, 2023-01-03:
Hmm, I'm having trouble understanding what you mean. Are you saying that what I'm describing already exists?

chrisdutz, 2023-01-03:
I think: yes ... because otherwise the component wouldn't have done anything before.

SteinOv, 2023-01-03:
Okay, thanks. I'll send an email to the mailing list to see if someone could help me in getting this to work.

SteinOv, 2023-01-04:
I just remembered this was stated in de docs: 
`The consumer supports one-time reading or Triggered Reading. (_Schedulded Reading using Period only soon_).`
If I understand correctly that means that what I am looking for is marked as "coming soon". Of course it could be that this has been added but the docs have not been updated, since the last update of the docs was in 2020.

SteinOv, 2023-01-19:
@chrisdutz
Hi Chris, I did not get a response from anyone after sending an email to the dev mailing list.
I think time based polling is not supported, see my previous comment. Are you okay with it if I implement it in the camel plc4x repository?
I could implement one of the two ways I specified in my original post. But I also thought about a third option; I could implement it that you have to configure the trigger with a certain string ("timed") for example and the period can be used to specify the polling interval.
What do you think?

chrisdutz, 2023-01-20:
Well I am not sure we want to continue maintaining two versions of the plugin ... My plans were actually to delete it from the repo here before the next release.

chrisdutz, 2023-01-20:
I'll contact the folks on the camel list again ... and see if I can have them have a look ... I would be happy to do so, but I don't have commit privileges there.

chrisdutz, 2023-01-20:
I'm actually not seeing your email to the camel list ... could you send me a lists.apache.org link to it, so I can refer to it?

https://lists.apache.org/list.html?dev@camel.apache.org

SteinOv, 2023-01-20:
I sent an email on the plc4x mailing list, since the Camel component was maintained here until recently. So, if anyone would know about the existing trigger functionality they would most likely be on the plc4x mailing list instead of the Camel mailing list. This is the email I sent: https://lists.apache.org/thread/str5qm2br0gqxzcsjycslc76x49tysp2.

With the camel plc4x repository I meant the plc4x component in the Camel repository. I can make a pull request over there for the changes, but I wanted you to agree first. Since last time you said you thought the plc4x component already supported a time based trigger. And I'd like to know which implementation you would prefer.

chrisdutz, 2023-01-20:
Well ... you would need to create a PR for the camel repo ... then I could comment on it there.

chrisdutz, 2023-01-20:
As far as I understood it, it's camel asking the component to produce an "event" ... so the triggering is done from Camel and not PLC4X ... not sure if there are pushing camel components.

SteinOv, 2023-01-20:
Okay, I'll look into it next week.