apache/plc4x #722: [Feature Request]: is there any example about plc4x-server and plc4-driver?

kdxq, 2022-12-31:
### What would you like to happen?

I have been tracking the new features of the plc4x. I like it. The new version has the plc4x-server, but I haven't seen any related examples. Can you add an example for it? Thank you very much .

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

apoorv28goel, 2023-01-19:
I am also looking for the same. It is very difficult to use the server without any kind of documentation. i have created a docker file for the PLC4X-Server but I am not able to test it. Can someone explain how to use the PLC4X-Server ?

chrisdutz, 2023-01-19:
Ok ... the PLC4X Server is sort of an experiment ... should be pretty advanced state, if you use the Java Server.

From what I see in the code I wrote some time ago, you simply start the Java PLC4X Server ... there's not any configuration to that. 

On the client side things are slightly more complicated.

Connection String: 

`plc4x://{ip-of-the-plc4x-server}?remote-connection-string={url-encoded-connection-string}`

I think you simply use the normal address strings you would use in order to communicate with the corresponding driver if you connected to it directly.

But ... as I said ... this thing is not very highly tested. But feel free to do so and give us some feedback on how it's performing and mabe help us make it better.

apoorv28goel, 2023-01-19:
thanks a lot @chrisdutz . Can you give more examples of requests like how to browse , read and write to a device ? .I intend to make the server production grade. I will try to contribute where ever I can. like documenting the requests, and fixing bugs.

chrisdutz, 2023-01-19:
I'm not sure Discovery, Subscription and Browse are implemented and tested at all ...

apoorv28goel, 2023-01-19:
i can see read and write are implemented. how to do I use those ? and example ?

chrisdutz, 2023-01-19:
Yeah ... so if you ask the connection about it's features, it says "false" for subscriptions and browsing. Probably not too hard to add, if you want to give it a try?

chrisdutz, 2023-01-19:
I posed in my response, how you build the connection url and that you simply use the nomal address strings (like for local versions of the driver)

apoorv28goel, 2023-01-19:
alright i can give it a try. do you mean like "plc4x://{ip-of-the-plc4x-server}?{tag-address1,tag-address2} " to read 2 tags ?

chrisdutz, 2023-01-19:
No ... like:
`plc4x://{ip-of-the-plc4x-server}?remote-connection-string={url-encoded-connection-string}`
And then using normal address strings for reading / writing.

Example (s7):
`plc4x://192.168.23.200?remote-connection-string=s7%3A%2F%2F192.168.23.30`

And then reading the addresses:
%DB4:4:DWORD, %DB4:22:DINT, ... or whatever you want to read.

closed 2023-09-25 by kdxq