apache/plc4x #731: [Feature Request]: Can the s7-200-cn plc support it?

nmgwr, 2023-01-06:
### What would you like to happen?

Can the s7-200-cn plc support it?
[image]

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

nmgwr, 2023-01-06:
[image]

chrisdutz, 2023-01-06:
Well ... so far we haven't implemented a S7 protocol via serial ... in theory it should be possible to do so, but I would say that out-of-the-box the current driver doesn't support this.

nmgwr, 2023-01-06:
https://github.com/s7connector/s7connector
I tried to use this driver to read and write, but there are bugs in this driver

nmgwr, 2023-01-06:
[image]
type is S200

nmgwr, 2023-01-06:
If you want to support the s7 200 cn in the future, I hope I can help you!

nmgwr, 2023-01-06:
The s7 200 smart is indeed supported by the test of the development branch I used. S7 200 cn reports the following error：

java.io.IOException: Connection reset by peer
	at java.base/sun.nio.ch.FileDispatcherImpl.read0(Native Method) ~[na:na]
	at java.base/sun.nio.ch.SocketDispatcher.read(SocketDispatcher.java:39) ~[na:na]
	at java.base/sun.nio.ch.IOUtil.readIntoNativeBuffer(IOUtil.java:276) ~[na:na]
	at java.base/sun.nio.ch.IOUtil.read(IOUtil.java:233) ~[na:na]
	at java.base/sun.nio.ch.IOUtil.read(IOUtil.java:223) ~[na:na]
	at java.base/sun.nio.ch.SocketChannelImpl.read(SocketChannelImpl.java:356) ~[na:na]
	at io.netty.buffer.PooledByteBuf.setBytes(PooledByteBuf.java:258) ~[netty-buffer-4.1.77.Final.jar:4.1.77.Final]
	at io.netty.buffer.AbstractByteBuf.writeBytes(AbstractByteBuf.java:1132) ~[netty-buffer-4.1.77.Final.jar:4.1.77.Final]
	at io.netty.channel.socket.nio.NioSocketChannel.doReadBytes(NioSocketChannel.java:357) ~[netty-transport-4.1.77.Final.jar:4.1.77.Final]
	at io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:151) ~[netty-transport-4.1.77.Final.jar:4.1.77.Final]
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:722) ~[netty-transport-4.1.77.Final.jar:4.1.77.Final]
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:658) ~[netty-transport-4.1.77.Final.jar:4.1.77.Final]
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:584) ~[netty-transport-4.1.77.Final.jar:4.1.77.Final]
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:496) ~[netty-transport-4.1.77.Final.jar:4.1.77.Final]
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:995) ~[netty-common-4.1.77.Final.jar:4.1.77.Final]
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74) ~[netty-common-4.1.77.Final.jar:4.1.77.Final]
	at io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30) ~[netty-common-4.1.77.Final.jar:4.1.77.Final]
	at java.base/java.lang.Thread.run(Thread.java:829) ~[na:na]

org.apache.plc4x.java.api.exceptions.PlcConnectionException: org.apache.plc4x.java.api.exceptions.PlcConnectionException: java.util.concurrent.ExecutionException: org.apache.plc4x.java.api.exceptions.PlcIoException: Connection terminated by remote
	at org.apache.plc4x.java.utils.connectionpool.PooledPlcDriverManager.getConnection(PooledPlcDriverManager.java:122)
	at org.apache.plc4x.java.utils.connectionpool.PooledPlcDriverManager.getConnection(PooledPlcDriverManager.java:101)
	at com.nmgwr.plc4xdemo.demo.Demo.run(Demo.java:33)
	at org.springframework.boot.SpringApplication.callRunner(SpringApplication.java:762)
	at org.springframework.boot.SpringApplication.callRunners(SpringApplication.java:752)
	at org.springframework.boot.SpringApplication.run(SpringApplication.java:315)
	at org.springframework.boot.SpringApplication.run(SpringApplication.java:1306)
	at org.springframework.boot.SpringApplication.run(SpringApplication.java:1295)
	at com.nmgwr.plc4xdemo.Plc4xDemoApplication.main(Plc4xDemoApplication.java:10)
Caused by: org.apache.plc4x.java.api.exceptions.PlcConnectionException: java.util.concurrent.ExecutionException: org.apache.plc4x.java.api.exceptions.PlcIoException: Connection terminated by remote
	at org.apache.plc4x.java.spi.connection.DefaultNettyPlcConnection.connect(DefaultNettyPlcConnection.java:144)
	at org.apache.plc4x.java.PlcDriverManager.getConnection(PlcDriverManager.java:78)
	at org.apache.plc4x.java.utils.connectionpool.PooledPlcDriverManager.access$1(PooledPlcDriverManager.java:1)
	at org.apache.plc4x.java.utils.connectionpool.PooledPlcDriverManager$1.create(PooledPlcDriverManager.java:90)
	at org.apache.plc4x.java.utils.connectionpool.PooledPlcDriverManager$1.create(PooledPlcDriverManager.java:1)
	at org.apache.commons.pool2.BaseKeyedPooledObjectFactory.makeObject(BaseKeyedPooledObjectFactory.java:82)
	at org.apache.commons.pool2.impl.GenericKeyedObjectPool.create(GenericKeyedObjectPool.java:780)
	at org.apache.commons.pool2.impl.GenericKeyedObjectPool.borrowObject(GenericKeyedObjectPool.java:439)
	at org.apache.commons.pool2.impl.GenericKeyedObjectPool.borrowObject(GenericKeyedObjectPool.java:350)
	at org.apache.plc4x.java.utils.connectionpool.PooledPlcDriverManager.getConnection(PooledPlcDriverManager.java:116)
	... 8 more
Caused by: java.util.concurrent.ExecutionException: org.apache.plc4x.java.api.exceptions.PlcIoException: Connection terminated by remote
	at java.base/java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:395)
	at java.base/java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1999)
	at org.apache.plc4x.java.spi.connection.DefaultNettyPlcConnection.connect(DefaultNettyPlcConnection.java:135)
	... 17 more
Caused by: org.apache.plc4x.java.api.exceptions.PlcIoException: Connection terminated by remote
	at org.apache.plc4x.java.spi.connection.DefaultNettyPlcConnection.lambda$1(DefaultNettyPlcConnection.java:127)
	at io.netty.util.concurrent.DefaultPromise.notifyListener0(DefaultPromise.java:578)
	at io.netty.util.concurrent.DefaultPromise.notifyListeners0(DefaultPromise.java:571)
	at io.netty.util.concurrent.DefaultPromise.notifyListenersNow(DefaultPromise.java:550)
	at io.netty.util.concurrent.DefaultPromise.notifyListeners(DefaultPromise.java:491)
	at io.netty.util.concurrent.DefaultPromise.setValue0(DefaultPromise.java:616)
[... cut ...]
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:658)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:584)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:496)
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:995)
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	at io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
	at java.base/java.lang.Thread.run(Thread.java:829)

chrisdutz, 2023-01-06:
Well the "host" and "port" on the above code snippet implicates a network connection ... the picture you sent indicates a serial communication. And yes... from a quick look at the "open issues" that project does seem to have quit a number of unattended issues ... I assume it's abandoned.

And regarding "me wanting to support" ... Today I finally burried my company (c-ware) ... My plans were to make a living based on PLC4X consulting, but the industry wasn't willing to pay for anything. So I have shifted my priorities. I do on PLC4X, what I need for my dayjob at Rivian or what I'm simply interested in. 

I am interested in helping the community and the project alive, but I'm no longer simply implementing features, just because they were asked for. 

So I am happy to assist you with implementing this feature, but I'm not going to do it for you. (Mabe someone else will do that)

chrisdutz, 2023-01-06:
If you want to use the S7200 smart thing ... you need to tell the driver about the type of plc, so it will not ask the PLC. Because in case of the S7200 it just hangs up (which you are seeing in your exception) ... so please add: 
`?controller-type=S7_200`
To your connection string.

nmgwr, 2023-01-06:
String connectionString = "s7://192.168.33.203?remote-rack=0&remote-slot=1&controller-type=S7_200";

this is my connectionString.

Anyway, I would like to thank you for your reply. Plc4x is a great project! I will use it in my project. As for s7 200 cn, I will think of other ways. Thank you.

closed 2023-01-06 by nmgwr

john1337, 2025-09-06:
> controller-type=S7_200

it doesn't work in my way,same exception thrown as nmgwr described

chrisdutz, 2025-09-06:
It's hard for us to test and improve, as we Noone I know has one of those devices.

john1337, 2025-09-06:
> It's hard for us to test and improve, as we Noone I know has one of those devices.

i do have this s7-200,device, when i am connecting to s7-200,follow exception came out,can u give me any clue to find the reeaseon? :
org.apache.plc4x.java.api.exceptions.PlcConnectionException: java.util.concurrent.ExecutionException: org.apache.plc4x.java.api.exceptions.PlcIoException: Connection terminated by remote
	at org.apache.plc4x.java.spi.connection.DefaultNettyPlcConnection.connect(DefaultNettyPlcConnection.java:157)
	at org.apache.plc4x.java.DefaultPlcDriverManager.getConnection(DefaultPlcDriverManager.java:80)
	at com.nimbus.LabPlcTcpDemo.main(LabPlcTcpDemo.java:39)
Caused by: java.util.concurrent.ExecutionException: org.apache.plc4x.java.api.exceptions.PlcIoException: Connection terminated by remote
	at java.base/java.util.concurrent.CompletableFuture.reportGet(CompletableFuture.java:395)
	at java.base/java.util.concurrent.CompletableFuture.get(CompletableFuture.java:1999)
	at org.apache.plc4x.java.spi.connection.DefaultNettyPlcConnection.connect(DefaultNettyPlcConnection.java:148)
	... 2 more
Caused by: org.apache.plc4x.java.api.exceptions.PlcIoException: Connection terminated by remote
	at org.apache.plc4x.java.spi.connection.DefaultNettyPlcConnection.lambda$1(DefaultNettyPlcConnection.java:140)
	at io.netty.util.concurrent.DefaultPromise.notifyListener0(DefaultPromise.java:603)
	at io.netty.util.concurrent.DefaultPromise.notifyListeners0(DefaultPromise.java:596)
	at io.netty.util.concurrent.DefaultPromise.notifyListenersNow(DefaultPromise.java:572)
	at io.netty.util.concurrent.DefaultPromise.notifyListeners(DefaultPromise.java:505)
	at io.netty.util.concurrent.DefaultPromise.setValue0(DefaultPromise.java:649)
	at io.netty.util.concurrent.DefaultPromise.setSuccess0(DefaultPromise.java:638)
	at io.netty.util.concurrent.DefaultPromise.trySuccess(DefaultPromise.java:118)

chrisdutz, 2025-09-06:
Some devices don't support the messages that we use to identify the type of device and simply hang up (logo devices so the same). If you pass in the type of controller, these messages are skipped and the driver works... So try passing in the s7-200