apache/plc4x #785: how to set rack and slot in plc4c

DamonDBT, 2023-02-06:
### What would you like to happen?

i use plc4c to connect S7-400,
according to the example :
[https://github.com/apache/plc4x/blob/develop/plc4c/examples/hello-world-s7/src/hello_world_s7_loopback.c#L34](url)
i did not find how to set rack and slot .

in plc4j we  use the fellow way, i tried in plc4c ,but failed.
[https://github.com/apache/plc4x/blob/develop/plc4j/examples/plc4j-s7event/src/main/java/org/apache/plc4x/examples/plc4j/s7event/EventSubscription.java#L45](url)

would please show me how to set rack and slot in plc4c ?
thanks.

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

kdxq, 2023-02-06:
[image]
i try it like this.but failed. @chrisdutz ,can you give some suggest? thank you very much

DamonDBT, 2023-02-06:
@kdxq thanks for your answer。
it works well.

closed 2023-02-06 by DamonDBT