apache/plc4x #791: [Bug]:

kdxq, 2023-02-08:
### What happened?

I'm writing a program in plc4c for use in an industrial production environment. Now I have a problem. When there are many tags in a request, there will be no response. I have encountered a similar situation when I used the early version of plc4j, maybe it is pdu and division? Please give some advice, thanks a lot. @christ 

I used wireshark to capture packets, but I couldn't upload the relevant attachments because github does not support the relevant attachment types, the screenshot is as follows. Let me know if any information is needed and I will do my best to provide it

[image]

### Version

v.0.10.0

### Programming Languages

- [ ] plc4j
- [ ] plc4go
- [X] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

chrisdutz, 2023-02-08:
PLC4C was always intended to be a super small implementation of PLC4X drivers, aimed at super tiny embedded computers. For the sake of the size of these drivers, the PLC4X drivers don't have the same set of features the full blown drivers in Java and Go have. 

One of the features the C version is missing, is the query-optimizer, that takes care of transparently splitting up large plc4x requests into multiple smaller S7 requests. How this splitting is done, depends on the type of S7 and the corresponding PDU size. 

When using PLC4C you will have to manually ensure the PDU size of your PLC is not exceeded manually and you'll have to execute multiple separate requests.

chrisdutz, 2023-02-08:
if you're not targeting super tiny embedded devices, I would strongly recommend to have a look at PLC4Go.
Alternatively we are also starting to work on PLC4Rust, but for that not a single driver is yet available. But if you're willing to help implement, there's a lot that can be speed up.

chrisdutz, 2023-02-08:
Well ... I can't say what PLC4X is planning as we are a community. I can just speak for myself, that I'm not planning on doing. it. So for now: I guess splitting up is the way to go for you ... or investigate how to call Go code from C++.

chrisdutz, 2023-02-08:
Well ... as you can imagine ... implementing PLC drivers in C is sort of nothing for the faint of heart. 

Everything is a lot more work in C. 

I implemented this because the EU was financing this work. 

I will not implement features like this, which others need for their production systems for free in my free time ... so unless someone else is willing to help, you plan on doing it or you or your company are willing to pay for it, I'm not seeing this feature coming any time soon.

kdxq, 2023-02-09:
I retested it. The data of one request are far less than 222 byte, but there will still be similar errors in wireshark, which should not be the problem of pdu.  @chrisdutz

kdxq, 2023-02-09:
Hi, @chrisdutz 
I tested with different data types, such as bool, int and real, and found that if the number of tags is greater than or equal to 40, there will be no response. If it is a pdu-related exception, the number of tags should be related to the data type. For example, the number of boot variables should be more than some and will not be unresponsive. Am I right? Can you give some suggestions? As you said yesterday, I want to implement the manual split request，Thank you.

chrisdutz, 2023-02-09:
Well the size of the cotp packet die both the request and the matching response is not allowed to exceed that size.

That's why die example you can't treat even one string without explicitly setting the size, because then the return string would be 255 bytes long and already exceed the size.

kdxq, 2023-02-09:
@chrisdutz 
 Thank you very much.

kdxq, 2023-02-10:
Hi,@chrisdutz
If the PLC data is read in the form of subscription, only when the first subscription is sent request to the PLC? In the following time, is the PLC actively sending data? Can subscription form reduce the number of requests? Am I right?

kdxq, 2023-02-10:
Hi,@chrisdutz
another question.
Can the plc4j achieve the 10ms timed data acquisition task? 
plc4j plc4c plc4go... Is the efficiency of collecting data the same  with the same hardware conditions?

thomas169, 2023-02-10:
https://github.com/apache/plc4x/blob/develop/plc4c/examples/hello-world-s7/src/hello_world_s7_loopback.c
I think this will test loopback time on Linux, as I used PLC4c to read/write to PLC for hardware in the loop type stuff. 
With solid hardware in testing I was able to get ~1ms time. Efficiency of this is 99% depending on hardware, unless you deliberately write code otherwise. Saying that I had started (unfinished) work to make sending/reading the same things repeatedly more deterministic  (less `alloc` use).

chrisdutz, 2023-02-10:
But this tests the Java driver, right? I think the author is explicitly referring to the C version, right? 
I haven't done any analysis, but I would guess it should be able to stay below 10ms for a single request/reply roundtrip. 
However I never tested this. So please ... just try it out.

thomas169, 2023-02-10:
I was referring to C also,  it all I ever used in PLC4x.

chrisdutz, 2023-02-10:
Oh ... I'm so sorry ... didn't even notice we had that example in there ;-)

heyoulin, 2023-02-14:
Try my resp [https://github.com/spnettec/plc4x.git](https://github.com/apache/plc4x/issues/url)? heyoulin branch. I run 2000 tags and no problem

kdxq, 2023-02-17:
hi @heyoulin thanks for your help. i will try it latter. Have you implemented an optimizer for plc4c？

chrisdutz, 2023-02-17:
Hi all ... I would like to say that I'm not too happy with this approach, where @heyoulin forks our repo, changes stuff and then guides people to his repo from the official one. 

@heyoulin it would be awesome, if you would start to creating PRs, so we can possibly integrate these changes back ... because this is not the way open-source should be done.

heyoulin, 2023-02-17:
> 

@chrisdutz I will create some PRs as soon as posible. 
Because many changes i have done. Some changes need many types of PLCs  for verification. Some are experimental modifications. I need time to separate then to independent PR. This will  take a lot of my time.
 I don't mean to guides people to my repo. I have no so many PLCs. I need some people help me to verify using real PLC. 
I want submit a PR after verifying that the PLC is executing correctly.

dremsol, 2023-02-18:
@heyoulin I can give you remote access to a S7300 CPU & S71500 CPU if you like.

spnettec, 2023-02-19:
> @heyoulin I can give you remote access to a S7300 CPU & S71500 CPU if you like. We could define an address space consisting of different data types and sizes to use for testing.

@dremsol Thank you. Could you send me the remote ip and address space via email?   heyoulin@gmail.com.

kdxq, 2023-03-10:
hi @heyoulin  @spnettec , I tested your branch. When the number of tags is greater than 40, no data is returned, and all variable types are plclist, and the printed values are bool array, could you give me some suggest? thank you very much.

chrisdutz, 2023-09-22:
So ... is @heyoulin "taking care of this issue"? I would say, that it's actually not a bug, but "works as designed" ... can we close this issue?

closed 2023-09-25 by kdxq