apache/plc4x #801: [Bug]: S7 driver - java.lang.OutOfMemoryError: unable to create native thread

DmitriiMukhin, 2023-02-13:
### What happened?

When s7 driver can't establish connection with controller it creates threads that being falling into park state TIMED_WAITING. While application keep trying to connect via PlcDriverManger.getConnection(url) such threads count continuously growing. In the end we have java.lang.OutOfMemoryError: unable to create native thread.

We are using s7 driver to communicate with different controllers from single app.

Screenshots just to demonstrate an issue. Driver creates thousands of such threads.

[image]
[image]
[image]
![Screenshot from 2023-02-13 16-10-05](https://user-images.githubusercontent.com/88187872/218467095-a7f7e4a5-c
[image]
d5d-4777-8305-2b8dcaa51a98.png)

### Version

v0.10.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

heyoulin, 2023-02-13:
Can you try my resp [https://github.com/spnettec/plc4x.git](url)? If no problem i will cretae a pull request.

DmitriiMukhin, 2023-02-13:
@heyoulin 
Thank you - great job.
I think this part of your solution helped me. 
```
                pipeline.addLast(new ChannelInboundHandlerAdapter() {
                                     @Override
                                     public void exceptionCaught(ChannelHandlerContext ctx, Throwable cause)
                                             throws PlcConnectionException {
                                         logger.error("unknown error, close the connection", cause);
                                         close();
                                     }
                                 }
                );
```
I've tested for a while - at first glance it works. Abandoned(parked) threads are absent. Connection is establishing as it should in normal situation, I mean when controller is accessible. I'll continue testing.

I'll be forced to fork 0.10.0 release to our company repo with such fix. As we don't know when new fixed release will be published.

DmitriiMukhin, 2023-02-14:
> @heyoulin Thank you - great job. I think this part of your solution helped me.
>
> ```
>                 pipeline.addLast(new ChannelInboundHandlerAdapter() { [...]

I was wrong. As I see only this part needed to fix that issue.

```
                if (!awaitSessionSetupComplete) {
                    setProtocol(
                            stackConfigurer.configurePipeline(
                                    configuration,
                                    pipeline,
                                    getAuthentication(),
                                    channelFactory.isPassive()
                            )
                    );
                }
```

DmitriiMukhin, 2023-02-14:
> @heyoulin Thank you - great job. I think this part of your solution helped me.
>
> ```
>                 pipeline.addLast(new ChannelInboundHandlerAdapter() { [...]

I was wrong. All code changes are needed.

chrisdutz, 2023-09-22:
Is this issue still open or has it been fixed?

DmitriiMukhin, 2023-09-25:
> Is this issue still open or has it been fixed?

It could be fixed with provided PR. Such fix was tested in our local company fork. No more issues.

chrisdutz, 2023-09-25:
Ok ... as I've merged that PR, I'll close this issue, right?

closed 2023-09-25 by chrisdutz