apache/plc4x #831: [Bug]: WARN org.apache.plc4x.java.spi.GeneratedDriverByteToMessageCodec - Error decoding package with content [010314127000000000126a000000000000000000000000c9]: Index 24 out of bounds for length 24

StrawberryBlue, 2023-03-04:
### What happened?

`String connectionString = "modbus-rtu:serial://COM2?unit-identifier=1&baudRate=9600&stopBits=1&parityBits=None&dataBits=8";
        PlcConnection plcConnection = new PlcDriverManager().getConnection(connectionString);
        PlcReadRequest.Builder builder = plcConnection.readRequestBuilder();
        builder.addItem("value-1", "40001[10]");

        PlcReadRequest build = builder.build();

        while (true){
            PlcReadResponse readResponse = build.execute().get(200, TimeUnit.MILLISECONDS);

            for (String fieldName : readResponse.getFieldNames()) {
                if (readResponse.getResponseCode(fieldName) == PlcResponseCode.OK){
                    int numValues = readResponse.getNumberOfValues(fieldName);
                    // If it's just one element, output just one single line.
                    if(numValues == 1) {
                        System.out.println("Value[" + fieldName + "]: " + readResponse.getObject(fieldName));
                    }
                    // If it's more than one element, output each in a single row.
                    else {
                        System.out.println("Value[" + fieldName + "]:");
                        for(int i = 0; i < numValues; i++) {
                            System.out.println(" - " + readResponse.getObject(fieldName, i));
                        }
                    }
                }
                // Something went wrong, to output an error message instead.
                else {
                    System.out.println("Error[" + fieldName + "]: " + readResponse.getResponseCode(fieldName).name());
                }
            }
            System.out.println("*******************************************************************************");
            Thread.sleep(100);
        }`

this code with 
`17:13:54.529 [nioEventLoopGroup-2-1] WARN org.apache.plc4x.java.spi.GeneratedDriverByteToMessageCodec - Error decoding package with content [010314127000000000126a000000000000000000000000c9]: Index 24 out of bounds for length 24
java.lang.ArrayIndexOutOfBoundsException: Index 24 out of bounds for length 24
	at com.github.jinahya.bit.io.ArrayByteInput.read(ArrayByteInput.java:106)
	at com.github.jinahya.bit.io.DefaultBitInput.read(DefaultBitInput.java:57)
	at com.github.jinahya.bit.io.AbstractBitInput.unsigned8(AbstractBitInput.java:63)
	at com.github.jinahya.bit.io.AbstractBitInput.unsigned16(AbstractBitInput.java:87)
	at com.github.jinahya.bit.io.AbstractBitInput.readInt(AbstractBitInput.java:130)
	at org.apache.plc4x.java.spi.generation.ReadBufferByteBased.readUnsignedInt(ReadBufferByteBased.java:175)
	at org.apache.plc4x.java.spi.codegen.io.DataReaderSimpleUnsignedInt.read(DataReaderSimpleUnsignedInt.java:33)
	at org.apache.plc4x.java.spi.codegen.io.DataReaderSimpleUnsignedInt.read(DataReaderSimpleUnsignedInt.java:25)
	at org.apache.plc4x.java.spi.codegen.fields.FieldReaderChecksum.lambda$readChecksumField$0(FieldReaderChecksum.java:36)
	at org.apache.plc4x.java.spi.codegen.FieldCommons.switchParseByteOrderIfNecessary(FieldCommons.java:69)
	at org.apache.plc4x.java.spi.codegen.fields.FieldReaderChecksum.readChecksumField(FieldReaderChecksum.java:36)
	at org.apache.plc4x.java.spi.codegen.fields.FieldReaderFactory.readChecksumField(FieldReaderFactory.java:81)
	at org.apache.plc4x.java.modbus.readwrite.ModbusRtuADU.staticParseBuilder(ModbusRtuADU.java:139)
	at org.apache.plc4x.java.modbus.readwrite.ModbusADU.staticParse(ModbusADU.java:121)
	at org.apache.plc4x.java.modbus.readwrite.ModbusADU.staticParse(ModbusADU.java:106)
	at org.apache.plc4x.java.modbus.rtu.ModbusRtuDriver.lambda$getStackConfigurer$0(ModbusRtuDriver.java:107)
	at org.apache.plc4x.java.spi.GeneratedDriverByteToMessageCodec.decode(GeneratedDriverByteToMessageCodec.java:87)
	at io.netty.handler.codec.ByteToMessageCodec$1.decode(ByteToMessageCodec.java:42)
	at io.netty.handler.codec.ByteToMessageDecoder.decodeRemovalReentryProtection(ByteToMessageDecoder.java:529)
	at io.netty.handler.codec.ByteToMessageDecoder.callDecode(ByteToMessageDecoder.java:468)
	at io.netty.handler.codec.ByteToMessageDecoder.channelRead(ByteToMessageDecoder.java:290)
	at io.netty.handler.codec.ByteToMessageCodec.channelRead(ByteToMessageCodec.java:103)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:442)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:412)
	at io.netty.channel.DefaultChannelPipeline$HeadContext.channelRead(DefaultChannelPipeline.java:1410)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:440)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
	at io.netty.channel.DefaultChannelPipeline.fireChannelRead(DefaultChannelPipeline.java:919)
	at org.apache.plc4x.java.transport.serial.SerialChannel$SerialNioUnsafe.read(SerialChannel.java:289)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:788)
[... 139 lines cut ...]
### Version

v0.10.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [X] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2024-02-10:
I've updated this driver quite a lot. Could you please try if the current 0.12.0-SNAPSHOT works?

chrisdutz, 2024-05-24:
I've just pushed an update that should make Modbus ASCII and RTU a bit more stable in situations as you describe. Please give the SNAPSHOT a try as soon as Jenkins is finished building it.

https://ci-builds.apache.org/job/PLC4X/job/PLC4X/job/develop/2050/

closed 2024-05-28 by StrawberryBlue