apache/plc4x #838: [Feature Request]:  can PLC4J fetch  data more faster?

kdxq, 2023-03-11:
### What would you like to happen?

hi,@chrisdutz. I am  using plc4j to collect data from S-400. At present, the 400 tag acquisition cycle can reach 100-120ms. The data types include INT, REAL, BOOL. .I hope the faster the better，especially for the bool variable. Can you give some suggestions？ thankyou.

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

glcj, 2023-03-11:
Hello,

I assume the S-400 is a Simatic S7-400.

Due to the design characteristics of these CPUs you will not be able to obtain that level of data acquisition that really represents your data, you must also analyze the execution cycles of your program OB1 (typical 1 to 10 ms), OB35 (100 - 500 ms typical), etc. and its technological functions (PCS7, Braumat, etc.), this is based on the fact that the data in real time is not the fastest, but the one that really represents the state of the variable.

In the world of DCS with a second everyone is happy (well almost). In the world of Safety Instrumented Systems (SIS) 1 millisecond is the norm.

If you really need a really fast variable update, like in PCS7 security systems, you can follow the guidelines in [1], where it sets out the software and hardware requirements.

You can also investigate the use of ALARM_8, ALARM_8P, for data that must be accurate depending on the execution of your program.

My grain of sand,

1.https://cache.industry.siemens.com/dl/files/111/68154111/att_80204/v1/ps7timeb_en-US.pdf

chrisdutz, 2023-09-22:
I guess we can close this issue, right?

closed 2023-09-25 by kdxq