apache/plc4x #849: [Bug]: plc4x-proxy mode error

kdxq, 2023-03-17:
### What happened?

Hi, @chrisdutz,  I plan to use the plc4x proxy mode, but I encountered an error during the testing process. When this error occurs several times, it will be automatically interrupted.detail information as follow. thank you very much!

plc4x-server just start your program. and 
plc4x-driver code adn error info. as follow. and the driver can received the data  from server. 
-----------------------------------------------------------------
    public static void main(String[] args) throws Exception {

        new Thread(() -> {
            while (true) {
                FetchTask();
                try {
                    Thread.sleep(2000);
                } catch (InterruptedException e) {
                    throw new RuntimeException(e);
                }
            }
        }).start();

    }

 private static void FetchTask(){
        String c_string ="plc4x://192.168.43.122?remote-connection-string=s7%3A%2F%2F10.32.137.171?remote-rack=0&remote-slot=3";
        PlcDriverManager manager = new PlcDriverManager();
        try (final PlcConnection connection =manager.getConnection(c_string)) {
            final PlcReadRequest.Builder requestBuilder = connection.readRequestBuilder();
            requestBuilder.addItem("Seg1_Cy1_Act_Power", "%DB101.DBD456:REAL");
            final PlcReadRequest readRequest = requestBuilder.build();
            final PlcReadResponse readResponse = readRequest.execute().get();
            float seg1Cy1ActPower = readResponse.getPlcValue("Seg1_Cy1_Act_Power").getFloat();
//            System.out.println("abc "+ seg1Cy1ActPower);
        } catch (Exception e) {
            throw new RuntimeException(e);
        }
    }

ERROR INFO AS FOLLOW:
------------------------------------------------------------------
Mar 17, 2023 11:34:28 AM io.netty.channel.DefaultChannelPipeline onUnhandledInboundException
WARNING: An exceptionCaught() event was fired, and it reached at the tail of the pipeline. It usually means the last handler in the pipeline did not handle the exception.
io.netty.handler.codec.DecoderException: java.lang.IllegalArgumentException: Unknown Transaction or Transaction already finished!
	at io.netty.handler.codec.MessageToMessageDecoder.channelRead(MessageToMessageDecoder.java:98)
	at io.netty.handler.codec.MessageToMessageCodec.channelRead(MessageToMessageCodec.java:111)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:357)
	at io.netty.handler.codec.ByteToMessageDecoder.fireChannelRead(ByteToMessageDecoder.java:336)
	at io.netty.handler.codec.ByteToMessageDecoder.channelRead(ByteToMessageDecoder.java:308)
	at io.netty.handler.codec.ByteToMessageCodec.channelRead(ByteToMessageCodec.java:103)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:357)
	at io.netty.channel.DefaultChannelPipeline$HeadContext.channelRead(DefaultChannelPipeline.java:1410)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	at io.netty.channel.DefaultChannelPipeline.fireChannelRead(DefaultChannelPipeline.java:919)
	at io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:166)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:788)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:724)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:650)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:562)
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:997)
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	at io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.IllegalArgumentException: Unknown Transaction or Transaction already finished!
	at org.apache.plc4x.java.spi.transaction.RequestTransactionManager.endRequest(RequestTransactionManager.java:129)
	at org.apache.plc4x.java.spi.transaction.RequestTransactionManager.access$200(RequestTransactionManager.java:46)
	at org.apache.plc4x.java.spi.transaction.RequestTransactionManager$RequestTransaction.endRequest(RequestTransactionManager.java:160)
	at org.apache.plc4x.java.plc4x.protocol.Plc4xProtocolLogic.lambda$read$7(Plc4xProtocolLogic.java:126)
	at org.apache.plc4x.java.spi.Plc4xNettyWrapper.decode(Plc4xNettyWrapper.java:187)
	at io.netty.handler.codec.MessageToMessageCodec$2.decode(MessageToMessageCodec.java:81)
	at io.netty.handler.codec.MessageToMessageDecoder.channelRead(MessageToMessageDecoder.java:88)
	... 23 more

### Version

v.0.10.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

kdxq, 2023-03-17:
and now the plc4x proxy mode not support  "CachedDriverManager" ? am  I right?

chrisdutz, 2023-09-22:
Sorry for the late response ... guess I'll have to dig into this one ...

chrisdutz, 2023-09-22:
A quick look at your code shows me, you're opening and closing the connection on every request. I would strongly suggest not to do that ... cause you're not only setting up and closing the connection on the PLC4X side, but the remote will have to establish and teardown the connection to the S7 too ... so I would strongly reccommend to keep the connection open or to use a cached connection manager. But still I'll be looking into this issue.

chrisdutz, 2024-08-16:
Closing for lack of activity.

closed 2024-08-16 by chrisdutz