apache/plc4x #866: Unable to start PLC4X route in camel-plc4x

mdanish98, 2023-03-25:
I am using [ModbusPal](https://plc4x.apache.org/users/getting-started/virtual-modbus.html) to create the Virtual modbus.

I am using the endpoint URI as:
`plc4x:modbus-tcp://localhost:502?unitId=1&dataType=holding-register&addresses=1`

Following is the screenshot of Modbus holding registers:
[image]

But when starting camel context it throws NullPointerException.
```
java.lang.NullPointerException
    at org.apache.camel.component.plc4x.Plc4XConsumer.startUnTriggered(Plc4XConsumer.java:89)
    at org.apache.camel.component.plc4x.Plc4XConsumer.doStart(Plc4XConsumer.java:81)
    at org.apache.camel.support.service.BaseService.start(BaseService.java:119)
```

I debugged it and found the problem is with this line:
` for (Map.Entry<String, Object> tag : tags.entrySet()) {`

Here **tags** is null, so basically I am not configuring the endpoint correctly.

How can I write the endpoint correctly?

chrisdutz, 2023-09-22:
Even if we have moved the plc4x camel component to the Apache Camel project, I could possibly create a PR, if it involves the camel-integration ... could you please provide a bit more information on the issue you are having (like an example source)?

mdanish98, 2023-09-23:
Hi @chrisdutz ,

Thanks for asking.
Sorry, I forgot to update this issue. Actually, this is fixed from Apache Camel's side:
https://issues.apache.org/jira/browse/CAMEL-19199

closed 2023-09-23 by mdanish98