apache/plc4x #870: about ads Test

ppppoooo, 2023-03-28:
I wrote an acquisition program using plc4x's ads and would like to ask if there is an ads simulator or a test program generated from ads simulation data to test the correctness of my program.
I hope you can give me advice, thank you very much!

ppppoooo, 2023-03-28:
After getting the ads data plc4x will report an error, can you help to look at it?
`17:32:42.305 [nioEventLoopGroup-2-1] WARN org.apache.plc4x.java.spi.GeneratedDriverByteToMessageCodec - Error decoding package with content [000020000000c0a8a3090101bd80c0a8a3000001530309000500000000004000000001000000]: Index 38 out of bounds for length 38
java.lang.ArrayIndexOutOfBoundsException: Index 38 out of bounds for length 38
	at com.github.jinahya.bit.io.ArrayByteInput.read(ArrayByteInput.java:106)
	at com.github.jinahya.bit.io.DefaultBitInput.read(DefaultBitInput.java:57)
	at com.github.jinahya.bit.io.AbstractBitInput.unsigned8(AbstractBitInput.java:63)
	at com.github.jinahya.bit.io.AbstractBitInput.unsigned16(AbstractBitInput.java:92)
	at com.github.jinahya.bit.io.AbstractBitInput.readInt(AbstractBitInput.java:135)
	at com.github.jinahya.bit.io.AbstractBitInput.readInt(AbstractBitInput.java:118)
	at com.github.jinahya.bit.io.AbstractBitInput.readLong(AbstractBitInput.java:156)
	at org.apache.plc4x.java.spi.generation.ReadBufferByteBased.readUnsignedLong(ReadBufferByteBased.java:172)
	at org.apache.plc4x.java.ads.readwrite.io.AdsReadWriteResponseIO.staticParse(AdsReadWriteResponseIO.java:65)
	at org.apache.plc4x.java.ads.readwrite.io.AdsDataIO.staticParse(AdsDataIO.java:138)
	at org.apache.plc4x.java.ads.readwrite.io.AmsPacketIO.staticParse(AmsPacketIO.java:101)
	at org.apache.plc4x.java.ads.readwrite.io.AmsTCPPacketIO.staticParse(AmsTCPPacketIO.java:75)
	at org.apache.plc4x.java.ads.readwrite.io.AmsTCPPacketIO.parse(AmsTCPPacketIO.java:48)
	at org.apache.plc4x.java.ads.readwrite.io.AmsTCPPacketIO.parse(AmsTCPPacketIO.java:42)
	at org.apache.plc4x.java.spi.GeneratedDriverByteToMessageCodec.decode(GeneratedDriverByteToMessageCodec.java:79)
	at io.netty.handler.codec.ByteToMessageCodec$1.decode(ByteToMessageCodec.java:42)
	at io.netty.handler.codec.ByteToMessageDecoder.decodeRemovalReentryProtection(ByteToMessageDecoder.java:507)
	at io.netty.handler.codec.ByteToMessageDecoder.callDecode(ByteToMessageDecoder.java:446)
	at io.netty.handler.codec.ByteToMessageDecoder.channelRead(ByteToMessageDecoder.java:276)
	at io.netty.handler.codec.ByteToMessageCodec.channelRead(ByteToMessageCodec.java:103)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:357)
	at io.netty.channel.DefaultChannelPipeline$HeadContext.channelRead(DefaultChannelPipeline.java:1410)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	at io.netty.channel.DefaultChannelPipeline.fireChannelRead(DefaultChannelPipeline.java:919)
	at io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:166)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:719)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:655)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:581)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:493)
	at io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:986)
	at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	at io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
	at java.base/java.lang.Thread.run(Thread.java:829)
`

splatch, 2023-03-28:
Hello @ppppoooo,
The program you linked is an old symbol scanner I implemented and proved to work ~3 years ago. @chrisdutz ported it last year, so you can use AdsDiscoveryDriver he made as starting point.

Please note that symbol scanning logic may differ between TwinCAT 2 and TwinCAT 3 (that's where I got stuck 3 years ago). The index out of bounds error reminds me it could be related to how TC3 reported data. This issue is for sure gone in Chris work.

ppppoooo, 2023-04-04:
Hello @chrisdutz ,
Can you send me the link to the demo, thank you very much!

> This

splatch, 2023-04-04:
@ppppoooo In my earlier comment I wrote exact class name you need to look for. Please check present ads driver sources.

closed 2023-04-04 by splatch