apache/plc4x #929: [Bug]: Starting subscription error using 0.10.0 package

cptjackwu, 2023-05-05:
### What happened?

 <dependency>
            <groupId>org.apache.plc4x</groupId>
            <artifactId>plc4j-api</artifactId>
            <version>0.10.0</version>
        </dependency>

        <dependency>
            <groupId>org.apache.plc4x</groupId>
            <artifactId>plc4j-driver-s7</artifactId>
            <version>0.10.0</version>
            <scope>runtime</scope>
        </dependency>

Exception in thread "main" java.lang.ClassCastException: org.apache.plc4x.java.s7.readwrite.field.S7StringField cannot be cast to org.apache.plc4x.java.s7.readwrite.field.S7SubscriptionField
	at org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.subscribe(S7ProtocolLogic.java:341)
	at org.apache.plc4x.java.spi.optimizer.BaseOptimizer.lambda$optimizedSubscribe$4(BaseOptimizer.java:120)
	at org.apache.plc4x.java.spi.optimizer.BaseOptimizer.send(BaseOptimizer.java:138)
	at org.apache.plc4x.java.spi.optimizer.BaseOptimizer.optimizedSubscribe(BaseOptimizer.java:120)
	at org.apache.plc4x.java.spi.connection.AbstractPlcConnection.subscribe(AbstractPlcConnection.java:183)
	at org.apache.plc4x.java.spi.messages.DefaultPlcSubscriptionRequest.execute(DefaultPlcSubscriptionRequest.java:66)
	at PlcClient.main(PlcClient.java:127)

### Version

v0.10.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

cptjackwu, 2023-05-05:
Is it because my address was written incorrectly? But when I read it, writing like that can read the values in the PLC。
example:  %DB5000:5.1:BOOL

chrisdutz, 2023-05-05:
No ... not all protocols support subscriptions ... if you had asked the connection if subscriptions are supported, it would have said "no". In the future we are planning on simulating subsciptions by polling in the background, but this feature still needs to be implemented. So I would suggest, you use the Read-API instead.

cptjackwu, 2023-05-06:
Thank you. I hope you can monitor changes in PLC addresses in the near future

closed 2023-05-09 by cptjackwu

DamonDBT, 2023-06-14:
got same problem when use plc4j 0.10.0 to subscript to s7 PLC.
the fellow check  got **true**.
`plcConnection.getMetadata().canSubscribe() `

but when run the demo :
```
      final PlcSubscriptionRequest.Builder builder = plcConnection.subscriptionRequestBuilder();
      String[] address = new String[2];
      List<String[]> taglist = new ArrayList<>();
      address[0] = "DB145.DBD10";
      address[1] = "%DB145.DBD10:REAL";
      taglist.add(address);
      for (int i = 0; i < taglist.size(); i++) {
          builder.addChangeOfStateField(taglist.get(0)[0], taglist.get(0)[1]);
      }
      PlcSubscriptionRequest subscriptionRequest = builder.build();
      final PlcSubscriptionResponse subscriptionResponse = subscriptionRequest.execute().get();
```
still got error :  org.apache.plc4x.java.s7.readwrite.field.S7Field cannot be cast to org.apache.plc4x.java.s7.readwrite.field.S7SubscriptionField

chrisdutz, 2023-06-14:
As we mentioned in previous responses ... a S7-1500 doesn't support subscriptions ... ony a very small subset of the S7 devices actually support it.

cptjackwu, 2023-06-14:
Due to the fact that many devices are now using S7-1500, we hope that PLC4J can achieve subscription

cptjackwu, 2023-06-14:
> Well this is actually not something that PLC4X can do ... the devices simply don't support it ... we could simulate it, but that comes at operational costs of higher load on the PLC. The S7 400 models that support it, do this via a separate mechanism, that is highly efficient and doesn't put too much extra load on the PLC.

Okay, I can only subscribe to the device myself. Thank you very much for your help.