apache/plc4x #930: [Feature Request]: S7_1500字符串可以读取，但是无法写入

cptjackwu, 2023-05-07:
### What would you like to happen?

[image]
[image]

When I write to the same address, I will report an error

[image]

[image]

So I'm not sure where the problem is right now. Can you help me point out the problem？

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

cptjackwu, 2023-05-07:
When I write like this, I also report the same error
 dataTowrite[1] =(byte)7;

hongjinlin, 2023-05-08:
Please provide the details of the error message.

cptjackwu, 2023-05-08:
[nioEventLoopGroup-2-1] INFO org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic - S7 Driver running in ACTIVE mode.

Exception in thread "main" java.nio.BufferOverflowException

at java.nio.Buffer.nextPutIndex(Buffer.java:521)

at java.nio.HeapByteBuffer.put(HeapByteBuffer.java:169)

at com.github.jinahya.bit.io.BufferByteOutput.write(BufferByteOutput.java:127)

at com.github.jinahya.bit.io.DefaultBitOutput.write(DefaultBitOutput.java:56)

at com.github.jinahya.bit.io.AbstractBitOutput.unsigned8(AbstractBitOutput.java:72)

at com.github.jinahya.bit.io.AbstractBitOutput.unsigned16(AbstractBitOutput.java:94)

at com.github.jinahya.bit.io.AbstractBitOutput.writeInt(AbstractBitOutput.java:120)

at com.github.jinahya.bit.io.AbstractBitOutput.writeByte(AbstractBitOutput.java:106)

at org.apache.plc4x.java.spi.generation.WriteBufferByteBased.writeSignedByte(WriteBufferByteBased.java:211)

at org.apache.plc4x.java.spi.generation.WriteBufferByteBased.writeByteArray(WriteBufferByteBased.java:103)

at org.apache.plc4x.java.spi.generation.WriteBuffer.writeByteArray(WriteBuffer.java:48)

at org.apache.plc4x.java.s7.readwrite.utils.StaticHelper.serializeS7String(StaticHelper.java:2047)

at org.apache.plc4x.java.s7.readwrite.DataItem.staticSerialize(DataItem.java:349)

at org.apache.plc4x.java.s7.readwrite.DataItem.staticSerialize(DataItem.java:248)

at org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.serializePlcValue(S7ProtocolLogic.java:865)

at org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.write(S7ProtocolLogic.java:287)

at org.apache.plc4x.java.spi.optimizer.BaseOptimizer.lambda$optimizedWrite$2(BaseOptimizer.java:113)

at org.apache.plc4x.java.spi.optimizer.BaseOptimizer.send(BaseOptimizer.java:150)

at org.apache.plc4x.java.spi.optimizer.BaseOptimizer.optimizedWrite(BaseOptimizer.java:113)

at org.apache.plc4x.java.spi.connection.AbstractPlcConnection.write(AbstractPlcConnection.java:175)

at org.apache.plc4x.java.spi.messages.DefaultPlcWriteRequest.execute(DefaultPlcWriteRequest.java:66)

at Write.main(Write.java:51)

cptjackwu, 2023-05-08:
String  value = "ac";
        byte[] stringvalue = value.getBytes();
        byte[] dataTowrite = new byte[stringvalue.length+2];
        dataTowrite[0] = (byte)2;  
        dataTowrite[1] =(byte)2; 
        System.arraycopy(stringvalue,0 ,dataTowrite,2,stringvalue.length);

builder.addItem("value-3", "%DB5000:6:STRING(2)",dataTowrite);
or
builder.addItem("value-3", "%DB5000:6:STRING",dataTowrite);
or
builder.addItem("value-3", "%DB5000:6:STRING(2)",stringvalue );
or
builder.addItem("value-3", "%DB5000:6:STRING(2)",value);

......

I have tried writing n types and all reported the same error。

spnettec, 2023-07-10:
Could you try my repository?

chrisdutz, 2023-07-10:
@spnettec: Please stop simply blindly guiding folks to your fork ... this is totally not a cool way to participate in an open-source project. If you find bugs, it would be nice if you would create PRs that address one issue at a time and if you would start accepting a discussion around your solution that goes beyond: "That's what I needed". Working on a project like PLC4X requires considering more than that one use-case you have. If you continue simply replying with "use my fork" on every issue here, I'll start removing them.

hongjinlin, 2023-07-10:
@spnettec I totally agree with Chris, this is the second time we've brought this up. Thanks anyway for your contribution to PLC4X.

astecker, 2023-09-22:
As a workaround you could try to write the String as multiple USINT (as i had problems writing them as an array):

val db = 1000
val lengthOfStringInS7 = 30
val string: String = "abc"
val toWrite = (""+Char(lengthOfStringInS7)+Char(string.length)+string).take(lengthOfStringInS7+2).toCharArray().mapIndexed{ idx, chr ->
            Triple("fancy_field_name_$idx", "${db+idx}:USINT" , chr.code)
}

  val writeResponse = with(plcConnection.writeRequestBuilder()) {
      toWrite.forEach { (name, fieldQuery, value) ->
          addItem(name, fieldQuery, value)
      }
      build()
  }.execute()[5000, TimeUnit.MILLISECONDS]

Code is Kotlin, and it's a stripped excerpt of a larger project, so expect some errors while substituting variable names. And better lookup the meaning of the first two bytes. For me it works, but it looks like its not checked by the S7 when written..?

chrisdutz, 2024-02-06:
Should be fixed now.

closed 2024-02-06 by chrisdutz