apache/plc4x #938: [Bug]:  S7_1500 String can be read, but cannot be written

cptjackwu, 2023-05-11:
### What happened?

https://github.com/apache/plc4x/issues/930#issue-1699086379

### Version

v0.10.0

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [X] S7

preveen-stack, 2023-05-14:
can you provide stack trace for this issue

cptjackwu, 2023-05-15:
2023-05-15 09:48:16.254  INFO   --- [                  main] org.apache.plc4x.java.PlcDriverManager   : Instantiating new PLC Driver Manager with class loader sun.misc.Launcher$AppClassLoader@18b4aac2
2023-05-15 09:48:16.267  INFO   --- [                  main] org.apache.plc4x.java.PlcDriverManager   : Registering available drivers...
2023-05-15 09:48:16.278  INFO   --- [                  main] org.apache.plc4x.java.PlcDriverManager   : Registering driver for Protocol s7 (Siemens S7 (Basic))
2023-05-15 09:48:16.878  INFO   --- [                  main] o.a.p.j.transport.tcp.TcpChannelFactory  : Configuring Bootstrap with Configuration{local-rack=1, local-slot=1, local-tsap=0, remote-rack=0, remote-slot=0, remote-tsap=0, pduSize=1024, maxAmqCaller=8, maxAmqCallee=8, controllerType='null'}
2023-05-15 09:48:17.574  INFO   --- [ nioEventLoopGroup-2-1] o.a.p.j.s.r.protocol.S7ProtocolLogic     : S7 Driver running in ACTIVE mode.
2023-05-15 09:48:17.728 ERROR   --- [                  main] com.wh.subscription.PlcClient            : Write Error: {}

java.nio.BufferOverflowException: null
	at java.nio.Buffer.nextPutIndex(Buffer.java:521)
	at java.nio.HeapByteBuffer.put(HeapByteBuffer.java:169)
	at com.github.jinahya.bit.io.BufferByteOutput.write(BufferByteOutput.java:127)
	at com.github.jinahya.bit.io.DefaultBitOutput.write(DefaultBitOutput.java:56)
	at com.github.jinahya.bit.io.AbstractBitOutput.unsigned8(AbstractBitOutput.java:72)
	at com.github.jinahya.bit.io.AbstractBitOutput.unsigned16(AbstractBitOutput.java:94)
	at com.github.jinahya.bit.io.AbstractBitOutput.writeInt(AbstractBitOutput.java:120)
	at com.github.jinahya.bit.io.AbstractBitOutput.writeByte(AbstractBitOutput.java:106)
	at org.apache.plc4x.java.spi.generation.WriteBufferByteBased.writeSignedByte(WriteBufferByteBased.java:211)
	at org.apache.plc4x.java.spi.generation.WriteBufferByteBased.writeByte(WriteBufferByteBased.java:97)
	at org.apache.plc4x.java.spi.generation.WriteBuffer.writeByte(WriteBuffer.java:42)
	at org.apache.plc4x.java.s7.readwrite.utils.StaticHelper.serializeS7String(StaticHelper.java:2046)
	at org.apache.plc4x.java.s7.readwrite.DataItem.staticSerialize(DataItem.java:349)
	at org.apache.plc4x.java.s7.readwrite.DataItem.staticSerialize(DataItem.java:248)
	at org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.serializePlcValue(S7ProtocolLogic.java:865)
	at org.apache.plc4x.java.s7.readwrite.protocol.S7ProtocolLogic.write(S7ProtocolLogic.java:287)
	at org.apache.plc4x.java.spi.optimizer.BaseOptimizer.lambda$optimizedWrite$2(BaseOptimizer.java:113)
	at org.apache.plc4x.java.spi.optimizer.BaseOptimizer.send(BaseOptimizer.java:150)
	at org.apache.plc4x.java.spi.optimizer.BaseOptimizer.optimizedWrite(BaseOptimizer.java:113)
	at org.apache.plc4x.java.spi.connection.AbstractPlcConnection.write(AbstractPlcConnection.java:175)
	at org.apache.plc4x.java.spi.messages.DefaultPlcWriteRequest.execute(DefaultPlcWriteRequest.java:66)
	at com.wh.subscription.PlcClient.execute(PlcClient.java:283)
	at WriteModel.main(WriteModel.java:19)

chrisdutz, 2023-07-21:
Is this issue still open?

chrisdutz, 2023-09-22:
I was able to verify it and did quite a number of updates in the Java S7 driver. Could you please give the current SNAPSHOT another try?

closed 2023-10-27 by cptjackwu