apache/plc4x #948: [Feature Request]: could ypu pls support semi hsms ?

shyding, 2023-05-16:
### What would you like to happen?

whan to support semi hsms ?

### Programming Languages

- [X] plc4j
- [ ] plc4go
- [ ] plc4c
- [ ] plc4net

### Protocols

- [ ] AB-Ethernet
- [ ] ADS /AMS
- [ ] BACnet/IP
- [ ] CANopen
- [ ] DeltaV
- [ ] DF1
- [ ] EtherNet/IP
- [ ] Firmata
- [ ] KNXnet/IP
- [ ] Modbus
- [ ] OPC-UA
- [ ] S7

chrisdutz, 2023-05-16:
I think we would need a bit more context here ... what are "semi hsms"? ... generally I am sure that we could. Problem is that usually for new drivers we would need hardware, software, specifications etc. and generally drivers are only implemented if someone does it himself or finances development in some form.

shyding, 2023-05-17:
exmple: https://github.com/kenta-shimizu/secs4java8.git

chrisdutz, 2023-09-22:
Well in general, we can support any protocol you throw at us ... the main problem probably is, that someone has to implement it. So if you're willing to get your hands dirty and start working on it the community will generally be happy to help. Or if you're willing to finance someone to implement it for you, that's fine too ... otherwise you'll have to wait till it lands on the plate of someone implementing it for you, but I guess none can give you an ETA on that.