apache/plc4x #952: Issue in running `docker build -t plc4x .`

maneetgoyal, 2023-05-17:
```
maneet@Maneets-MBP plc4x % docker build -t plc4x . 
[+] Building 721.5s (30/35)                                                                       
 => [build 12/24] RUN apt install -y dotnet-sdk-6.0                                         66.4s
 => [build 13/24] RUN apt install -y python3 python3-venv python3-pip                       31.1s
 => [build 14/24] RUN pip3 install wheel                                                     0.9s
 => [build 15/24] RUN apt install -y dos2unix                                                3.9s
 => [build 16/24] COPY . /ws/                                                                2.3s
 => [build 17/24] WORKDIR /ws                                                                0.0s
 => [build 18/24] RUN chmod +x ./mvnw                                                        0.2s
 => [build 19/24] RUN dos2unix ./mvnw                                            [+] Building 721.6s (30/35)                                                                        RUN dos2unix .mvn/wrapper/maven-wrapper.properties              => [build 12/24] RUN apt install -y dotnet-sdk-6.0                                         66.4s] RUN ./mvnw -P with-c,with-dotnet,with-go,with-sandbox com.offby => [build 13/24] RUN apt install -y python3 python3-venv python3-pip                       31.1srg/apache/maven/maven-compat/3.9.0/maven-compat-3.9.0.jar         => [build 14/24] RUN pip3 install wheel                                                     0.9sded from apache-releases: https://repository.apache.org/content/r => [build 15/24] RUN apt install -y dos2unix                                                3.9sg/apache/maven/maven-artifact/3.9.0/maven-artifact-3.9.0.jar (59  => [build 16/24] COPY . /ws/                                                                2.3sding from apache-releases: https://repository.apache.org/content/ => [build 17/24] WORKDIR /ws                                                                0.0srg/apache/maven/maven-model-builder/3.9.0/maven-model-builder-3.9 => [build 18/24] RUN chmod +x ./mvnw                                                        0.2ss (5): 0.3/1.0 MB | 208/217 kB | 463/502 kB | 179/290 kB | 172/19 => [build 19/24] RUN dos2unix ./mvnw                                            [+] Building 721.8s (30/35)                                                                        RUN dos2unix .mvn/wrapper/maven-wrapper.properties              => [build 12/24] RUN apt install -y dotnet-sdk-6.0                                         66.4s] RUN ./mvnw -P with-c,with-dotnet,with-go,with-sandbox com.offby => [build 13/24] RUN apt install -y python3 python3-venv python3-pip                       31.1srg/apache/maven/maven-compat/3.9.0/maven-compat-3.9.0.jar         => [build 14/24] RUN pip3 install wheel                                                     0.9sded from apache-releases: https://repository.apache.org/content/r => [build 15/24] RUN apt install -y dos2unix                                                3.9sg/apache/maven/maven-artifact/3.9.0/maven-artifact-3.9.0.jar (59  => [build 16/24] COPY . /ws/                                                                2.3sding from apache-releases: https://repository.apache.org/content/ => [build 17/24] WORKDIR /ws                                                                0.0srg/apache/maven/maven-model-builder/3.9.0/maven-model-builder-3.9 => [build 18/24] RUN chmod +x ./mvnw                                                        0.2ss (5): 0.3/1.0 MB | 212/217 kB | 467/502 kB | 187/290 kB | 172/19 => [build 19/24] RUN dos2unix ./mvnw                                            [+] Building 4927.8s (32/35)                                                     
 => [internal] load build definition from Dockerfile                        0.0s  => => transferring dockerfile: 37B                                         0.0s
 => [internal] load .dockerignore                                           0.0sy => => transferring context: 35B                                            0.0s
 => [internal] load metadata for docker.io/azul/zulu-openjdk:11             5.2s  => [internal] load metadata for docker.io/library/alpine:3.10              3.4s
 => [build  1/24] FROM docker.io/azul/zulu-openjdk:11@sha256:ae1836a7956a  18.4sr => => resolve docker.io/azul/zulu-openjdk:11@sha256:ae1836a7956a730892bb3  0.0s
 => => sha256:ae1836a7956a730892bb377ccbe864036c3c072d678e78da 770B / 770B  0.0s  => => sha256:7841b96d5fc54ddfdde57241cdf7dc95dfd0481cc76584a9 742B / 742B  0.0s
 => => sha256:4889516a6986fcea417446fba73337f899322b87859a 4.24kB / 4.24kB  0.0s/ => => sha256:a7016d2db01f67f2c18746e387008a1bb55b8b07 153.55MB / 153.55MB  7.1s
 => => extracting sha256:a7016d2db01f67f2c18746e387008a1bb55b8b076e2341e4  10.6s9 => [stage-2 1/3] FROM docker.io/library/alpine:3.10@sha256:451eee8bedcb2f  0.0s
 => [internal] load build context                                           2.7s9 => => transferring context: 5.01MB                                         2.4s
 => CACHED [packager 2/4] RUN apk update                                    0.0s
 => CACHED [packager 3/4] RUN apk --no-cache add openjdk11-jdk openjdk11-j  0.0s
 => CACHED [packager 4/4] RUN /usr/lib/jvm/java-11-openjdk/bin/jlink     -  0.0s
 => CACHED [stage-2 2/3] COPY --from=packager /opt/java-minimal /opt/java-  0.0s
 => [build  2/24] RUN apt update -y                                        71.8s
 => [build  3/24] RUN apt install -y make libpcap-dev libc-dev git         77.5s
 => [build  4/24] RUN apt install -y build-essential                      112.3s
 => [build  5/24] RUN apt install -y wget                                   4.9s
 => [build  6/24] RUN wget -q https://packages.microsoft.com/config/ubuntu  0.8s
 => [build  7/24] RUN dpkg -i packages-microsoft-prod.deb                   0.5s
 => [build  8/24] RUN apt install -y software-properties-common            52.8s
 => [build  9/24] RUN add-apt-repository universe -y                        5.3s
 => [build 10/24] RUN apt install -y apt-transport-https                    9.1s
 => [build 11/24] RUN apt update -y                                        10.6s
 => [build 12/24] RUN apt install -y dotnet-sdk-6.0                        66.4s
 => [build 13/24] RUN apt install -y python3 python3-venv python3-pip      31.1s
 => [build 14/24] RUN pip3 install wheel                                    0.9s
 => [build 15/24] RUN apt install -y dos2unix                               3.9s
 => [build 16/24] COPY . /ws/                                               2.3s
 => [build 17/24] WORKDIR /ws                                               0.0s
 => [build 18/24] RUN chmod +x ./mvnw                                       0.2s
 => [build 19/24] RUN dos2unix ./mvnw                                       0.2s
 => [build 20/24] RUN dos2unix .mvn/wrapper/maven-wrapper.properties        0.2s
 => [build 21/24] RUN ./mvnw -P with-c,with-dotnet,with-go,with-sandbox  4445.4s
 => ERROR [build 22/24] RUN ./mvnw -P with-c,with-dotnet,with-go,with-sand  7.6s
------
 > [build 22/24] RUN ./mvnw -P with-c,with-dotnet,with-go,with-sandbox install:
#32 1.440 [INFO] Scanning for projects...
#32 3.184 [INFO] ------------------------------------------------------------------------
#32 3.185 [INFO] Reactor Build Order:
#32 3.186 [INFO] 
#32 3.188 [INFO] PLC4X                                                              [pom]
#32 3.191 [INFO] Code-Generation                                                    [pom]
#32 3.193 [INFO] Code-Generation: Protocol Base: MSpec                              [jar]
#32 3.196 [INFO] Code-Generation: Language Base: Freemarker                         [jar]
#32 3.197 [INFO] PLC4J                                                              [pom]
#32 3.199 [INFO] PLC4J: API                                                         [jar]
#32 3.200 [INFO] PLC4J: SPI                                                         [jar]
#32 3.201 [INFO] PLC4J: Transports                                                  [pom]
#32 3.202 [INFO] PLC4J: Transports: TCP                                             [jar]
#32 3.203 [INFO] PLC4J: Transports: Test                                            [jar]
#32 3.204 [INFO] PLC4J: Utils                                                       [pom]
#32 3.204 [INFO] PLC4J: Utils: Test Utils                                           [jar]
[... cut ...]
#32 7.491 [ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
#32 7.492 [ERROR] Re-run Maven using the -X switch to enable full debug logging.
#32 7.492 [ERROR] 
#32 7.492 [ERROR] For more information about the errors and possible solutions, please read the following articles:
#32 7.492 [ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/PluginExecutionException
------
executor failed running [/bin/sh -c ./mvnw -P with-c,with-dotnet,with-go,with-sandbox install]: exit code: 1
maneet@Maneets-MBP plc4x % 
```

maneetgoyal, 2023-05-17:
cc @sruehl

sruehl, 2023-05-17:
ok seems like the error output is being eaten up. @maneetgoyal Try again please with: 320c7095c180c90e2c17caf033d124273d9b3846

ottobackwards, 2023-05-17:
I think the error is that with-dotnet is specified and it is not found isn't it?

sruehl, 2023-05-17:
yes, but I want to see the exact output of it (I assume you are on AMD64)

maneetgoyal, 2023-05-17:
```
maneet@Maneets-MBP plc4x % docker build -t plc4x . 
[+] Building 721.5s (30/35)                                                                       
 => [build 12/24] RUN apt install -y dotnet-sdk-6.0                                         66.4s
 => [build 13/24] RUN apt install -y python3 python3-venv python3-pip                       31.1s
 => [build 14/24] RUN pip3 install wheel                                                     0.9s
 => [build 15/24] RUN apt install -y dos2unix                                                3.9s
 => [build 16/24] COPY . /ws/                                                                2.3s
 => [build 17/24] WORKDIR /ws                                                                0.0s
 => [build 18/24] RUN chmod +x ./mvnw                                                        0.2s
 => [build 19/24] RUN dos2unix ./mvnw                                            [+] Building 721.6s (30/35)                                                                        RUN dos2unix .mvn/wrapper/maven-wrapper.properties              => [build 12/24] RUN apt install -y dotnet-sdk-6.0                                         66.4s] RUN ./mvnw -P with-c,with-dotnet,with-go,with-sandbox com.offby => [build 13/24] RUN apt install -y python3 python3-venv python3-pip                       31.1srg/apache/maven/maven-compat/3.9.0/maven-compat-3.9.0.jar         => [build 14/24] RUN pip3 install wheel                                                     0.9sded from apache-releases: https://repository.apache.org/content/r => [build 15/24] RUN apt install -y dos2unix                                                3.9sg/apache/maven/maven-artifact/3.9.0/maven-artifact-3.9.0.jar (59  => [build 16/24] COPY . /ws/                                                                2.3sding from apache-releases: https://repository.apache.org/content/ => [build 17/24] WORKDIR /ws                                                                0.0srg/apache/maven/maven-model-builder/3.9.0/maven-model-builder-3.9 => [build 18/24] RUN chmod +x ./mvnw                                                        0.2ss (5): 0.3/1.0 MB | 208/217 kB | 463/502 kB | 179/290 kB | 172/19 => [build 19/24] RUN dos2unix ./mvnw                                            [+] Building 721.8s (30/35)                                                                        RUN dos2unix .mvn/wrapper/maven-wrapper.properties              => [build 12/24] RUN apt install -y dotnet-sdk-6.0                                         66.4s] RUN ./mvnw -P with-c,with-dotnet,with-go,with-sandbox com.offby => [build 13/24] RUN apt install -y python3 python3-venv python3-pip                       31.1srg/apache/maven/maven-compat/3.9.0/maven-compat-3.9.0.jar         => [build 14/24] RUN pip3 install wheel                                                     0.9sded from apache-releases: https://repository.apache.org/content/r => [build 15/24] RUN apt install -y dos2unix                                                3.9sg/apache/maven/maven-artifact/3.9.0/maven-artifact-3.9.0.jar (59  => [build 16/24] COPY . /ws/                                                                2.3sding from apache-releases: https://repository.apache.org/content/ => [build 17/24] WORKDIR /ws                                                                0.0srg/apache/maven/maven-model-builder/3.9.0/maven-model-builder-3.9 => [build 18/24] RUN chmod +x ./mvnw                                                        0.2ss (5): 0.3/1.0 MB | 212/217 kB | 467/502 kB | 187/290 kB | 172/19 => [build 19/24] RUN dos2unix ./mvnw                                            [+] Building 4927.8s (32/35)                                                     
 => [internal] load build definition from Dockerfile                        0.0s  => => transferring dockerfile: 37B                                         0.0s
 => [internal] load .dockerignore                                           0.0sy => => transferring context: 35B                                            0.0s
 => [internal] load metadata for docker.io/azul/zulu-openjdk:11             5.2s  => [internal] load metadata for docker.io/library/alpine:3.10              3.4s
 => [build  1/24] FROM docker.io/azul/zulu-openjdk:11@sha256:ae1836a7956a  18.4sr => => resolve docker.io/azul/zulu-openjdk:11@sha256:ae1836a7956a730892bb3  0.0s
 => => sha256:ae1836a7956a730892bb377ccbe864036c3c072d678e78da 770B / 770B  0.0s  => => sha256:7841b96d5fc54ddfdde57241cdf7dc95dfd0481cc76584a9 742B / 742B  0.0s
 => => sha256:4889516a6986fcea417446fba73337f899322b87859a 4.24kB / 4.24kB  0.0s/ => => sha256:a7016d2db01f67f2c18746e387008a1bb55b8b07 153.55MB / 153.55MB  7.1s
 => => extracting sha256:a7016d2db01f67f2c18746e387008a1bb55b8b076e2341e4  10.6s9 => [stage-2 1/3] FROM docker.io/library/alpine:3.10@sha256:451eee8bedcb2f  0.0s
 => [internal] load build context                                           2.7s9 => => transferring context: 5.01MB                                         2.4s
 => CACHED [packager 2/4] RUN apk update                                    0.0s
 => CACHED [packager 3/4] RUN apk --no-cache add openjdk11-jdk openjdk11-j  0.0s
 => CACHED [packager 4/4] RUN /usr/lib/jvm/java-11-openjdk/bin/jlink     -  0.0s
 => CACHED [stage-2 2/3] COPY --from=packager /opt/java-minimal /opt/java-  0.0s
 => [build  2/24] RUN apt update -y                                        71.8s
[... cut ...]
#32 6.709 [ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
#32 6.710 [ERROR] Re-run Maven using the -X switch to enable full debug logging.
#32 6.710 [ERROR] 
#32 6.710 [ERROR] For more information about the errors and possible solutions, please read the following articles:
#32 6.711 [ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/PluginExecutionException
------
executor failed running [/bin/sh -c ./mvnw -P with-c,with-dotnet,with-go,with-sandbox install]: exit code: 1
maneet@Maneets-MBP plc4x % 
```

sruehl, 2023-05-17:
@chrisdutz can you try a build on your MBP AMD64?

maneetgoyal, 2023-05-17:
Trying this too now:

```
# Build everything with all tests
# RUN ./mvnw -P with-c,with-dotnet,with-go,with-sandbox install
RUN ./mvnw install
```

Here's my machine details:

```
Software:

    System Software Overview:

      System Version: macOS 13.2.1 (22D68)
      Kernel Version: Darwin 22.3.0
      Boot Volume: Macintosh HD

Hardware:

    Hardware Overview:

      Model Name: MacBook Pro
      Processor Name: 8-Core Intel Core i9
```

chrisdutz, 2023-05-17:
Oh ... I already did and it worked right up to the step where the example it wants to mount, doesn't exist anymore (So we need to fix this), but not seeing any problems with .Net.

sruehl, 2023-05-17:
try following:
go into the docker bash with `docker run -it azul/zulu-openjdk:11 /bin/bash`
then within execute those commands:
```
apt update -y
apt install -y wget
wget -q https://packages.microsoft.com/config/ubuntu/20.04/packages-microsoft-prod.deb -O packages-microsoft-prod.deb
dpkg -i packages-microsoft-prod.deb
apt install -y software-properties-common
add-apt-repository universe -y
apt install -y apt-transport-https
apt update -y
apt install -y dotnet-sdk-6.0
dotnet --version

```

chrisdutz, 2023-05-17:
Is there possibly anything related to export restictions or "firewalls"?

maneetgoyal, 2023-05-17:
> Trying this too now:
>
> ```
> # Build everything with all tests
> # RUN ./mvnw -P with-c,with-dotnet,with-go,with-sandbox install
> RUN ./mvnw install
> ```
>
> Here's my machine details:
>
> ```
> Software: [...]

Output:

```
maneet@Maneets-MBP plc4x % docker build -t plc4x .
[+] Building 950.3s (34/35)                                                      
 => [internal] load build definition from Dockerfile                        0.0s
 => => transferring dockerfile: 4.95kB                                      0.0s
 => [internal] load .dockerignore                                           0.0s
 => => transferring context: 35B                                            0.0s
 => [internal] load metadata for docker.io/azul/zulu-openjdk:11             2.8s
 => [internal] load metadata for docker.io/library/alpine:3.10              2.8s
 => [build  1/24] FROM docker.io/azul/zulu-openjdk:11@sha256:ae1836a7956a7  0.0s
 => [packager 1/4] FROM docker.io/library/alpine:3.10@sha256:451eee8bedcb2  0.0s
 => [internal] load build context                                           0.7s
 => => transferring context: 1.12MB                                         0.6s
 => CACHED [packager 2/4] RUN apk update                                    0.0s
 => CACHED [packager 3/4] RUN apk --no-cache add openjdk11-jdk openjdk11-j  0.0s
 => CACHED [packager 4/4] RUN /usr/lib/jvm/java-11-openjdk/bin/jlink     -  0.0s
 => CACHED [stage-2 2/3] COPY --from=packager /opt/java-minimal /opt/java-  0.0s
 => CACHED [build  2/24] RUN apt update -y                                  0.0s
 => CACHED [build  3/24] RUN apt install -y make libpcap-dev libc-dev git   0.0s
 => CACHED [build  4/24] RUN apt install -y build-essential                 0.0s
 => CACHED [build  5/24] RUN apt install -y wget                            0.0s
 => CACHED [build  6/24] RUN wget -q https://packages.microsoft.com/config  0.0s
 => CACHED [build  7/24] RUN dpkg -i packages-microsoft-prod.deb            0.0s
 => CACHED [build  8/24] RUN apt install -y software-properties-common      0.0s
 => CACHED [build  9/24] RUN add-apt-repository universe -y                 0.0s
 => CACHED [build 10/24] RUN apt install -y apt-transport-https             0.0s
 => CACHED [build 11/24] RUN apt update -y                                  0.0s
 => CACHED [build 12/24] RUN apt install -y dotnet-sdk-6.0                  0.0s
 => CACHED [build 13/24] RUN apt install -y python3 python3-venv python3-p  0.0s
 => CACHED [build 14/24] RUN pip3 install wheel                             0.0s
 => CACHED [build 15/24] RUN apt install -y dos2unix                        0.0s
 => CACHED [build 16/24] COPY . /ws/                                        0.0s
 => CACHED [build 17/24] WORKDIR /ws                                        0.0s
 => CACHED [build 18/24] RUN chmod +x ./mvnw                                0.0s
 => CACHED [build 19/24] RUN dos2unix ./mvnw                                0.0s
 => CACHED [build 20/24] RUN dos2unix .mvn/wrapper/maven-wrapper.propertie  0.0s
 => CACHED [build 21/24] RUN ./mvnw -P with-c,with-dotnet,with-go,with-san  0.0s
 => [build 22/24] RUN ./mvnw install                                      928.5s
 => [build 23/24] RUN ./mvnw org.apache.maven.plugins:maven-help-plugin:3  17.9s
 => ERROR [build 24/24] RUN PROJECT_VERSION=`cat project_version`; mv plc4  0.3s
------
 > [build 24/24] RUN PROJECT_VERSION=`cat project_version`; mv plc4j/examples/hello-integration-iotdb/target/plc4j-examples-hello-integration-iotdb-${PROJECT_VERSION}-uber-jar.jar plc4xdemo.jar:
#34 0.224 mv: cannot stat 'plc4j/examples/hello-integration-iotdb/target/plc4j-examples-hello-integration-iotdb-0.11.0-SNAPSHOT-uber-jar.jar': No such file or directory
------
executor failed running [/bin/sh -c PROJECT_VERSION=`cat project_version`; mv plc4j/examples/hello-integration-iotdb/target/plc4j-examples-hello-integration-iotdb-${PROJECT_VERSION}-uber-jar.jar plc4xdemo.jar]: exit code: 1
```

maneetgoyal, 2023-05-17:
> @maneetgoyal what does [this](https://github.com/apache/plc4x/issues/952#issuecomment-1551189278) output for you?

Checking it shortly, got pulled onto something

maneetgoyal, 2023-05-17:
@sruehl Getting:

```
root@1c6855b4f5ca:/# apt install -y dotnet-sdk-6.0
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
dotnet-sdk-6.0 is already the newest version (6.0.408-1).
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
root@1c6855b4f5ca:/# dotnet --version
A fatal error occurred. The folder [/usr/share/dotnet/host/fxr] does not exist
```

Ran all the commands but shared the output of final 2

sruehl, 2023-05-17:
hmmm might be related to this https://stackoverflow.com/a/73899341

Strange that this doesn't happen for @chrisdutz

ottobackwards, 2023-05-17:
Is he using Intel or M1/2?  maybe that is a difference?
The user is using intel.

sruehl, 2023-05-17:
I told him to check on AMD64 but maybe he missed that...

maneetgoyal, 2023-05-18:
Output of `docker build -t plc4x` from an Apple M2 Machine:

```
#0 1059.6 [INFO] PLC4J: Transports: PCAP Replay ..................... SUCCESS [  0.404 s]
#0 1059.6 [INFO] PLC4J: Driver: S7 (Step7) .......................... SUCCESS [  6.893 s]
#0 1059.6 [INFO] PLC4J: Utils: Test-Generator ....................... SUCCESS [  3.998 s]
#0 1059.6 [INFO] PLC4J: Driver: BACNet/IP Driver .................... SUCCESS [ 30.717 s]
#0 1059.6 [INFO] PLC4J: Driver: C-Bus ............................... SUCCESS [  8.986 s]
#0 1059.6 [INFO] PLC4J: Transports: CAN ............................. SUCCESS [  0.283 s]
#0 1059.6 [INFO] PLC4J: Transports: SocketCAN ....................... SUCCESS [  0.819 s]
#0 1059.6 [INFO] PLC4J: Transports: VirtualCAN ...................... SUCCESS [  0.287 s]
#0 1059.6 [INFO] PLC4J: Driver: CAN ................................. SUCCESS [  1.926 s]
#0 1059.6 [INFO] PLC4J: Driver: CANopen ............................. SUCCESS [  3.562 s]
#0 1059.6 [INFO] PLC4J: Driver: Ethernet IP ......................... SUCCESS [  5.103 s]
#0 1059.6 [INFO] PLC4J: Driver: Firmata ............................. SUCCESS [  2.652 s]
#0 1059.6 [INFO] PLC4J: Driver: KNXNet/IP ........................... SUCCESS [  7.259 s]
#0 1059.6 [INFO] PLC4J: Driver: Mock ................................ SUCCESS [  0.373 s]
#0 1059.6 [INFO] PLC4J: Driver: Open-Protocol ....................... SUCCESS [  5.677 s]
#0 1059.6 [INFO] PLC4J: Driver: Modbus .............................. SUCCESS [  4.973 s]
#0 1059.6 [INFO] PLC4J: Driver: OPC UA .............................. SUCCESS [ 43.720 s]
#0 1059.6 [INFO] PLC4J: Driver: PLC4X (Proxy-Protocol) .............. SUCCESS [  2.419 s]
#0 1059.6 [INFO] PLC4J: Driver: Profinet ............................ SUCCESS [  5.628 s]
#0 1059.6 [INFO] PLC4J: Driver: Simulated ........................... SUCCESS [  7.297 s]
#0 1059.6 [INFO] PLC4J: Examples .................................... SUCCESS [  1.862 s]
#0 1059.6 [INFO] PLC4J: Examples: Hello-ADS telemetry ............... SUCCESS [  1.916 s]
#0 1059.6 [INFO] PLC4J: Examples: Cloud: Azure IoT Hub .............. SUCCESS [ 15.997 s]
#0 1059.6 [INFO] PLC4J: Examples: Cloud: Google Cloud IoT Core ...... SUCCESS [  7.914 s]
#0 1059.6 [INFO] PLC4J: Examples: Hello Discovery ................... SUCCESS [  1.871 s]
#0 1059.6 [INFO] PLC4J: Examples: Connectivity: MQTT ................ SUCCESS [  2.449 s]
#0 1059.6 [INFO] PLC4J: Examples: Hello InfluxDB Data Collection .... SUCCESS [  2.414 s]
#0 1059.6 [INFO] PLC4J: Tools ....................................... SUCCESS [  0.025 s]
#0 1059.6 [INFO] PLC4J: Tools: Object PLC Mapping ................... SUCCESS [  5.930 s]
#0 1059.6 [INFO] PLC4J: Tools: Connection Cache ..................... SUCCESS [  1.914 s]
#0 1059.6 [INFO] PLC4J: Examples: Hello OPM ......................... SUCCESS [  2.134 s]
#0 1059.6 [INFO] PLC4J: Examples: Hello-Webservice .................. SUCCESS [  1.953 s]
#0 1059.6 [INFO] PLC4J: Examples: Hello-World: Kotlin ............... SUCCESS [  7.143 s]
#0 1059.6 [INFO] PLC4J: Examples: Hello-World PLC4X (Discover And Browse) SUCCESS [  1.836 s]
#0 1059.6 [INFO] PLC4J: Examples: Hello-World PLC4X (Read) .......... SUCCESS [  1.869 s]
#0 1059.6 [INFO] PLC4J: Examples: Hello-World PLC4X (Subscribe) ..... SUCCESS [  1.864 s]
#0 1059.6 [INFO] PLC4J: Examples: Hello-World PLC4X (Write) ......... SUCCESS [  1.856 s]
#0 1059.6 [INFO] PLC4J: Examples: Poll Loop Example ................. SUCCESS [  1.839 s]
#0 1059.6 [INFO] PLC4J: Examples: S7Event example ................... SUCCESS [  1.850 s]
#0 1059.6 [INFO] PLC4J: Tools: Scraper .............................. SUCCESS [  4.090 s]
#0 1059.6 [INFO] PLC4J: Integrations ................................ SUCCESS [  0.026 s]
#0 1059.6 [INFO] PLC4J: Integrations: Apache Calcite ................ SUCCESS [  5.186 s]
#0 1059.6 [INFO] PLC4J: Integrations: Apache Kafka .................. SUCCESS [01:54 min]
#0 1059.6 [INFO] PLC4J: Integrations: Apache Nifi ................... SUCCESS [  0.849 s]
#0 1059.6 [INFO] PLC4J: Integrations: Apache Nifi: Processors ....... SUCCESS [  4.310 s]
#0 1059.6 [INFO] PLC4J: Integrations: Apache Nifi: NAR .............. SUCCESS [01:22 min]
#0 1059.6 [INFO] PLC4J: Integrations: Eclipse Milo OPC UA Server .... SUCCESS [  3.638 s]
#0 1059.6 [INFO] PLC4J: Tools: Capture Replay ....................... SUCCESS [  0.445 s]
#0 1059.6 [INFO] PLC4J: Tools: Plc4x Server ......................... SUCCESS [  1.261 s]
#0 1059.6 [INFO] PLC4J: Tools: Scraper (NG) ......................... SUCCESS [  0.050 s]
#0 1059.6 [INFO] PLC4J: Tools: UI ................................... SUCCESS [  0.391 s]
#0 1059.6 [INFO] PLC4J: Utils: PLC-Simulator ........................ SUCCESS [  2.012 s]
#0 1059.6 [INFO] PLC4J: OSGi ........................................ SUCCESS [  0.286 s]
#0 1059.6 [INFO] PLC4C .............................................. SUCCESS [ 15.683 s]
#0 1059.6 [INFO] PLC4Go ............................................. FAILURE [03:42 min]
#0 1059.6 [INFO] PLC4Net ............................................ SKIPPED
#0 1059.6 [INFO] Sandbox ............................................ SKIPPED
#0 1059.6 [INFO] Sandbox: PLC4J: Driver: Discovery .................. SKIPPED
#0 1059.6 [INFO] Sandbox: Test Df1 Driver ........................... SKIPPED
#0 1059.6 [INFO] ------------------------------------------------------------------------
#0 1059.6 [INFO] BUILD FAILURE
[... cut ...]
#0 1059.6 [ERROR] After correcting the problems, you can resume the build with the command
#0 1059.6 [ERROR]   mvn <args> -rf :plc4go
------
Dockerfile:69
--------------------
  67 |     
  68 |     # Build everything with all tests
  69 | >>> RUN ./mvnw -P with-c,with-dotnet,with-go,with-sandbox install
  70 |     
  71 |     # Get the version of the project and save it in a local file on the container
--------------------
ERROR: failed to solve: process "/bin/sh -c ./mvnw -P with-c,with-dotnet,with-go,with-sandbox install" did not complete successfully: exit code: 1
```

maneetgoyal, 2023-05-18:
<details>
  <summary>More detailed error log for M2</summary>
  

  ### Error Log (Apple M2)
  ```
#0 1059.6 [INFO] === Failed
#0 1059.6 [INFO] === FAIL: internal/cbus TestReader_sendMessageOverTheWire/Send_message_which_responds_with_confirm_only (0.01s)
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Send "}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Sending request"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Sending message"}
#0 1059.6 [INFO] {"level":"debug","time":"2023-05-17T16:37:38Z","message":"Created request context\n╔═RequestContext/sendIdentifyRequestBefore═══════════════════╗\n║                          b1 true                           ║\n╚════════════════════════════════════════════════════════════╝"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"4 bytes encoded"}
#0 1059.6 [INFO]     Reader_test.go:1216: Waiting now
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Receive"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Receive"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Receive"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Peek 1 readable bytes (0 available bytes)"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Peek 1 readable bytes (1 available bytes)"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Receive"}
#0 1059.6 [INFO] {"level":"trace","byte":103,"time":"2023-05-17T16:37:38Z","message":"current byte"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Peek 2 readable bytes (2 available bytes)"}
#0 1059.6 [INFO] {"level":"debug","error":"Error while peeking: not enough bytes available","time":"2023-05-17T16:37:38Z","message":"Error filling buffer"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Buffer filled"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Receive"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Peek 1 readable bytes (0 available bytes)"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Peek 1 readable bytes (0 available bytes)"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Receive"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Peek 1 readable bytes (0 available bytes)"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"return number of readable bytes 0"}
#0 1059.6 [INFO] {"level":"debug","error":"Error while peeking: not enough bytes available","time":"2023-05-17T16:37:38Z","message":"Error filling buffer"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Buffer filled"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Peek 1 readable bytes (0 available bytes)"}
#0 1059.6 [INFO] {"level":"debug","error":"Error while peeking: not enough bytes available","time":"2023-05-17T16:37:38Z","message":"Error filling buffer"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Buffer filled"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"return number of readable bytes 0"}
#0 1059.6 [INFO] {"level":"trace","byte":46,"time":"2023-05-17T16:37:38Z","message":"current byte"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Peek 3 readable bytes (3 available bytes)"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Receive"}
#0 1059.6 [INFO] {"level":"trace","byte":13,"time":"2023-05-17T16:37:38Z","message":"current byte"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Peek 1 readable bytes (0 available bytes)"}
#0 1059.6 [INFO] {"level":"debug","error":"Error while peeking: not enough bytes available","time":"2023-05-17T16:37:38Z","message":"Error filling buffer"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Buffer filled"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"return number of readable bytes 0"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Nothing to read"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Nothing to read"}
#0 1059.6 [INFO] {"level":"debug","error":"Error while peeking: not enough bytes available","time":"2023-05-17T16:37:38Z","message":"Error filling buffer"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Buffer filled"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"return number of readable bytes 0"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Nothing to read"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"return number of readable bytes 0"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Nothing to read"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"Buffer filled"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"return number of readable bytes 4"}
#0 1059.6 [INFO] {"level":"trace","time":"2023-05-17T16:37:38Z","message":"4 bytes available in buffer"}
[... cut ...]
#0 1059.6 [ERROR]   mvn <args> -rf :plc4go
------
Dockerfile:69
--------------------
  67 |     
  68 |     # Build everything with all tests
  69 | >>> RUN ./mvnw -P with-c,with-dotnet,with-go,with-sandbox install
  70 |     
  71 |     # Get the version of the project and save it in a local file on the container
--------------------
ERROR: failed to solve: process "/bin/sh -c ./mvnw -P with-c,with-dotnet,with-go,with-sandbox install" did not complete successfully: exit code: 1
  ```
</details>

chrisdutz, 2023-05-18:
So I just added a bit of a tiny update that the example at the end actually outputs something ... but on my M2 Mac it works just fine.

maneetgoyal, 2023-05-21:
```
[+] Building 3188.3s (36/36) FINISHED                                              
 => [internal] load build definition from Dockerfile                          0.0s
 => => transferring dockerfile: 5.03kB                                        0.0s
 => [internal] load .dockerignore                                             0.0s
 => => transferring context: 35B                                              0.0s
 => [internal] load metadata for docker.io/azul/zulu-openjdk:11               0.0s
 => [internal] load metadata for docker.io/library/alpine:3.10                2.6s
 => [build  1/24] FROM docker.io/azul/zulu-openjdk:11                         0.0s
 => [packager 1/4] FROM docker.io/library/alpine:3.10@sha256:451eee8bedcb2f0  0.0s
 => [internal] load build context                                             2.5s
 => => transferring context: 1.31MB                                           2.4s
 => CACHED [packager 2/4] RUN apk update                                      0.0s
 => CACHED [packager 3/4] RUN apk --no-cache add openjdk11-jdk openjdk11-jmo  0.0s
 => CACHED [packager 4/4] RUN /usr/lib/jvm/java-11-openjdk/bin/jlink     --v  0.0s
 => CACHED [stage-2 2/3] COPY --from=packager /opt/java-minimal /opt/java-mi  0.0s
 => CACHED [build  2/24] RUN apt update -y                                    0.0s
 => CACHED [build  3/24] RUN apt install -y make libpcap-dev libc-dev git     0.0s
 => CACHED [build  4/24] RUN apt install -y build-essential                   0.0s
 => CACHED [build  5/24] RUN apt install -y wget                              0.0s
 => CACHED [build  6/24] RUN wget -q https://packages.microsoft.com/config/u  0.0s
 => CACHED [build  7/24] RUN dpkg -i packages-microsoft-prod.deb              0.0s
 => CACHED [build  8/24] RUN apt install -y software-properties-common        0.0s
 => CACHED [build  9/24] RUN add-apt-repository universe -y                   0.0s
 => CACHED [build 10/24] RUN apt install -y apt-transport-https               0.0s
 => CACHED [build 11/24] RUN apt update -y                                    0.0s
 => CACHED [build 12/24] RUN apt install -y dotnet-sdk-6.0                    0.0s
 => CACHED [build 13/24] RUN apt install -y python3 python3-venv python3-pip  0.0s
 => CACHED [build 14/24] RUN pip3 install wheel                               0.0s
 => CACHED [build 15/24] RUN apt install -y dos2unix                          0.0s
 => [build 16/24] COPY . /ws/                                                 3.7s
 => [build 17/24] WORKDIR /ws                                                 0.0s
 => [build 18/24] RUN chmod +x ./mvnw                                         0.3s
 => [build 19/24] RUN dos2unix ./mvnw                                         0.2s
 => [build 20/24] RUN dos2unix .mvn/wrapper/maven-wrapper.properties          0.2s
 => [build 21/24] RUN ./mvnw -P with-c,with-dotnet,with-go,with-sandbox c  1428.6s
 => [build 22/24] RUN ./mvnw install                                       1730.3s
 => [build 23/24] RUN ./mvnw org.apache.maven.plugins:maven-help-plugin:3.2  18.2s
 => [build 24/24] RUN PROJECT_VERSION=`cat project_version`; mv plc4j/exampl  0.4s
 => [stage-2 3/3] COPY --from=build /ws/plc4xdemo.jar /plc4xdemo.jar          0.1s
 => exporting to image                                                        0.2s
 => => exporting layers                                                       0.2s
 => => writing image sha256:738b7024f65ba4af9af8358ddf46797435dfa12bf94b34d2  0.0s
 => => naming to docker.io/library/plc4x 
```
@chrisdutz Works okay after puling in the latest updates. Modified the dockerfile too:

```
# Build everything with all tests
# RUN ./mvnw -P with-c,with-dotnet,with-go,with-sandbox install
RUN ./mvnw install
```

Running a test without this change shortly.

_(Intel Chip)_

maneetgoyal, 2023-05-21:
The dotnet issue is still there:

```
#32 6.493 
#32 6.500 Detecting Java version:    11.0.19        OK
#32 6.551 Detecting Dotnet version:  missing
#32 6.585 --- output of version `dotnet --version` command ---
#32 6.586 
#32 6.587 A fatal error occurred. The folder [/usr/share/dotnet/host/fxr] does not exist
#32 6.587 
```

I guess for my usecase, it's okay as long as `RUN ./mvnw install` works (which is the case right now).

chrisdutz, 2023-07-21:
So I've spent a lot of time fine-tuning the Docker setup with docker-compose.

Could you give the "docker compose up" now a try?

chrisdutz, 2023-09-22:
Is this issue still an issue, or was it fixed by my cleanup?

maneetgoyal, 2023-09-22:
Sorry, was occupied. Will try to test today.

maneetgoyal, 2023-09-23:
Running `docker compose up` on M2/16GB. Taking a looong time. Will post updates once it completes.

sruehl, 2023-09-25:
Is it still booting up? :D

chrisdutz, 2023-09-25:
Well on my device it failed because of a problem in the OPC-UA testsuite, but it didn't take unreasonably long .... of course, if you start with an empy "out/" directoy (Like I just did), it will have to download the internet first, but once that's done it doesn't take much longer ...

maneetgoyal, 2023-09-25:
Will try again once the OPC-UA fix is in. Apologies for not being able to share the test results. It kept running for long and then I had to give away the machine for some other task.

chrisdutz, 2023-09-25:
I did notice that there seem to be issues with OPC-UA tests in Docker ... locally the build runs fine, but it seems in Docker they fail. We have had numerous problems with running Milo (which is used as a test-server) in the past. The thing is, that we have an integration-testsuite system configured, that would be able to run tests without neding a server, but it seems noone is using it. Perhaps I should convert the tests into using that instead.

chrisdutz, 2024-02-10:
So can I close this issue? I have built PLC4X using Docker many times now and no longer encountered any issues with it. Obviously when running for the first time, this takes quite a while as it has to download the internet first, but the second time you run it, it should be a lot quicker.

maneetgoyal, 2024-02-11:
@chrisdutz Sounds good, sure can be closed.

closed 2024-02-11 by chrisdutz